ostat/gridfinity_extended_openscad #1: multiple undefined operations and unknown variables for gridfinity_baseplate.scad

ph15h, 2023-12-13:
Opening gridfinity_baseplate.scad in OpenSCAD outputs these errors. I have all .scad files on the main page as well as the entire modules folder as well

```WARNING: Ignoring unknown variable 'gridfinity_pitch' in file gridfinity_baseplate.scad, line 91
WARNING: undefined operation (undefined * number) in file gridfinity_baseplate.scad, line 91
WARNING: Ignoring unknown variable 'gridfinity_pitch' in file gridfinity_baseplate.scad, line 91
WARNING: undefined operation (undefined * number) in file gridfinity_baseplate.scad, line 91
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file gridfinity_baseplate.scad, line 91
WARNING: Ignoring unknown variable 'gridfinity_corner_radius' in file gridfinity_baseplate.scad, line 310
WARNING: Ignoring unknown variable 'gridfinity_pitch' in file gridfinity_baseplate.scad, line 315
WARNING: undefined operation (undefined / number) in file gridfinity_baseplate.scad, line 315
WARNING: undefined operation (undefined - undefined) in file gridfinity_baseplate.scad, line 315
WARNING: undefined operation (undefined - number) in file gridfinity_baseplate.scad, line 315
WARNING: undefined operation (-undefined) in file modules/gridfinity_modules.scad, line 441
WARNING: undefined operation (-undefined) in file modules/gridfinity_modules.scad, line 442
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file modules/gridfinity_modules.scad, line 443
WARNING: undefined operation (undefined - undefined) in file gridfinity_baseplate.scad, line 324
WARNING: undefined operation (number * undefined) in file gridfinity_baseplate.scad, line 324
WARNING: undefined operation (undefined - undefined) in file gridfinity_baseplate.scad, line 325
WARNING: undefined operation (number * undefined) in file gridfinity_baseplate.scad, line 325
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file gridfinity_baseplate.scad, line 326
WARNING: undefined operation (-undefined) in file modules/gridfinity_modules.scad, line 441
WARNING: undefined operation (number + undefined) in file modules/gridfinity_modules.scad, line 442
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file modules/gridfinity_modules.scad, line 443
WARNING: undefined operation (undefined - undefined) in file gridfinity_baseplate.scad, line 324
WARNING: undefined operation (number * undefined) in file gridfinity_baseplate.scad, line 324
WARNING: undefined operation (undefined - undefined) in file gridfinity_baseplate.scad, line 325
WARNING: undefined operation (number * undefined) in file gridfinity_baseplate.scad, line 325
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file gridfinity_baseplate.scad, line 326
WARNING: undefined operation (number + undefined) in file modules/gridfinity_modules.scad, line 441
WARNING: undefined operation (-undefined) in file modules/gridfinity_modules.scad, line 442
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file modules/gridfinity_modules.scad, line 443
WARNING: undefined operation (undefined - undefined) in file gridfinity_baseplate.scad, line 324
WARNING: undefined operation (number * undefined) in file gridfinity_baseplate.scad, line 324
WARNING: undefined operation (undefined - undefined) in file gridfinity_baseplate.scad, line 325
WARNING: undefined operation (number * undefined) in file gridfinity_baseplate.scad, line 325
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file gridfinity_baseplate.scad, line 326
WARNING: undefined operation (number + undefined) in file modules/gridfinity_modules.scad, line 441
WARNING: undefined operation (number + undefined) in file modules/gridfinity_modules.scad, line 442
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file modules/gridfinity_modules.scad, line 443
WARNING: undefined operation (undefined - undefined) in file gridfinity_baseplate.scad, line 324
WARNING: undefined operation (number * undefined) in file gridfinity_baseplate.scad, line 324
WARNING: undefined operation (undefined - undefined) in file gridfinity_baseplate.scad, line 325
WARNING: undefined operation (number * undefined) in file gridfinity_baseplate.scad, line 325
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file gridfinity_baseplate.scad, line 326```

FireplayerDB, 2023-12-29:
Same for me.

`
WARNING: Ignoring unknown variable 'gridfinity_pitch' in file gridfinity_baseplate.scad, line 91 
WARNING: undefined operation (undefined * number) in file gridfinity_baseplate.scad, line 91 
WARNING: Ignoring unknown variable 'gridfinity_pitch' in file gridfinity_baseplate.scad, line 91 
WARNING: undefined operation (undefined * number) in file gridfinity_baseplate.scad, line 91 
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file gridfinity_baseplate.scad, line 91 
WARNING: Ignoring unknown variable 'gridfinity_corner_radius' in file gridfinity_baseplate.scad, line 310 
WARNING: Ignoring unknown variable 'gridfinity_pitch' in file gridfinity_baseplate.scad, line 315 
WARNING: undefined operation (undefined / number) in file gridfinity_baseplate.scad, line 315 
WARNING: undefined operation (undefined - undefined) in file gridfinity_baseplate.scad, line 315 
WARNING: undefined operation (undefined - number) in file gridfinity_baseplate.scad, line 315 
WARNING: Ignoring unknown variable '$idx' in file gridfinity_baseplate.scad, line 322 
WARNING: Ignoring unknown variable '$idx' in file gridfinity_baseplate.scad, line 322 
WARNING: undefined operation (undefined - number) in file gridfinity_baseplate.scad, line 324 
WARNING: Ignoring unknown variable '$idx' in file gridfinity_baseplate.scad, line 324 
WARNING: undefined operation (number * undefined) in file gridfinity_baseplate.scad, line 324 
WARNING: undefined operation (undefined - number) in file gridfinity_baseplate.scad, line 325 
WARNING: Ignoring unknown variable '$idx' in file gridfinity_baseplate.scad, line 325 
WARNING: undefined operation (number * undefined) in file gridfinity_baseplate.scad, line 325 
WARNING: undefined operation (-undefined) in file gridfinity_modules.scad, line 441 
WARNING: undefined operation (-undefined) in file gridfinity_modules.scad, line 442 
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file modules/gridfinity_modules.scad, line 443 
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file gridfinity_baseplate.scad, line 326 
WARNING: undefined operation (-undefined) in file gridfinity_modules.scad, line 441 
WARNING: undefined operation (number + undefined) in file gridfinity_modules.scad, line 442 
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file modules/gridfinity_modules.scad, line 443 
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file gridfinity_baseplate.scad, line 326 
WARNING: undefined operation (number + undefined) in file gridfinity_modules.scad, line 441 
WARNING: undefined operation (-undefined) in file gridfinity_modules.scad, line 442 
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file modules/gridfinity_modules.scad, line 443 
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file gridfinity_baseplate.scad, line 326 
WARNING: undefined operation (number + undefined) in file gridfinity_modules.scad, line 441 
WARNING: undefined operation (number + undefined) in file gridfinity_modules.scad, line 442 
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file modules/gridfinity_modules.scad, line 443 
WARNING: Unable to convert translate([undef, undef, 0]) parameter to a vec3 or vec2 of numbers in file gridfinity_baseplate.scad, line 326
`

OpenSCAD Version: 2021.01
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 16 CPUs 31.87 GB RAM
User Agent: OpenSCAD/2021.01 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.4.0" 64bit

antoncl, 2024-01-06:
Undoing the out commenting worked for me. I do _not_ get any $idx error

ostat, 2024-01-10:
This should be fixed with #6c32e56.
Sorry for the delay, thanks for raising

closed 2024-01-10 by ostat