ostat/gridfinity_extended_openscad #3: Plate size cannot exceed 6 x 6

antoncl, 2024-01-06:
If any of the dimensions of a baseplate exceeds 6 then you get an: 

WARNING: Normalized tree is growing past 200000 elements. Aborting normalization.

WARNING: CSG normalization resulted in an empty tree

ostat, 2024-01-10:
I need to look at ways to reduce the element count (I think you can use the render command, but this can have other impacts on the preview).
One solution is to the number of elements in the options, the default is 100000 (which gives the 200000 elements error). I have mine set to 500000, which works fine
[image]

closed 2024-02-20 by ostat

ar1a, 2024-09-16:
this would be great to mention in the readme or somewhere. same with enabling fast-csg

ostat, 2024-09-16:
good idea, I have added this to issue #45 
I personally don't enable **fast-csg**, as i get inconsistent results.