ostat/gridfinity_extended_openscad #4: Wall pattern does not apply to dividers.

brotherssmith, 2024-01-08:
I don't see the setting listed in the documentation, but there is a checkbox in the program, and it seems to do nothing. Have tried changing various wall pattern and divider settings to no effect.

Unrelated - also having the missing variables issue for baseplate generation, restoring the old variable names remedied some issues, but the missing Sdix remains for me, and it won't generate a full grid unless in baseplate also a lid mode.

Feature request - A baseplate/lid that is a "lite" version, missing magnet holes for efficiency.

Thanks for all the work you've put into this already.

ostat, 2024-01-10:
>I don't see the setting listed in the documentation, but there is a checkbox in the program, and it seems to do nothing. Have tried changing various wall pattern and divider settings to no effect.
This was a bug introduced when I rewrote the new divider logic. Its been fixed, but its only setup to work for the **Vertical chambers** option at the moment. 
fixed in e16e486
[image]

>Unrelated - also having the missing variables issue for baseplate generation, restoring the old variable names remedied some issues, but the missing Sdix remains for me, and it won't generate a full grid unless in baseplate also a lid mode.
I believe this was covered in #1, its fixed now

>Feature request - A baseplate/lid that is a "lite" version, missing magnet holes for efficiency.
>Example here - https://www.thingiverse.com/thing:5975756
Moved out to its own issue. #5

natalietrinh102, 2024-05-22:
Are there plans to support horizontal dividers? Only supporting vertical dividers leads to some odd label positioning.

ostat, 2024-05-22:
> Are there plans to support horizontal dividers? Only supporting vertical dividers leads to some odd label positioning.  

We do support horizontal dividers. However we don't support the patterned wall on horizontal dividers. This is beachside I would need to deal with the intersection area when both are enabled. 
Not sure I follow how this impacts the labels?

natalietrinh102, 2024-05-24:
I meant support for patterned walls on horizontal dividers. The labels positions currently don't allow for you to build a model with the label on the short side while having a patterned wall:

[image]
This is what I'm intending to do - with a patterned wall in the middle

[image]
This is the orientation that allows for a patterned wall - but the label is now on a side that I won't use. (There is also some odd geometry on vertical chamber in this case)

For now I've just exported it into a CAD program and created the pattern myself, but adding it would be useful for me in the future.

ostat, 2024-05-24:
We can do that.

ostat, 2024-05-25:
Fixed in e92073d .

[image]

ostat, 2024-06-10:
this issue is resolved

closed 2024-06-10 by ostat