ostat/gridfinity_extended_openscad #6: Baseplate issues

elbelga, 2024-01-31:
I dont not why but baseplate is cut on both sides

[image]

elbelga, 2024-02-01:
I have the latest version 
[image]

I set the same config but it doesnt works properly
[image]

I dont know if it is related but I have this warnings in console

`
[WARNING: Ignoring unknown variable 'butterflyClipEnabled' in file gridfinity_baseplate.scad, line 191](https://github.com/ostat/gridfinity_extended_openscad/issues/191,/Users/belga/Documents/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[WARNING: Ignoring unknown variable 'butterflyClipSize' in file gridfinity_baseplate.scad, line 192](https://github.com/ostat/gridfinity_extended_openscad/issues/192,/Users/belga/Documents/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[WARNING: Ignoring unknown variable 'butterflyClipRadius' in file gridfinity_baseplate.scad, line 193](https://github.com/ostat/gridfinity_extended_openscad/issues/193,/Users/belga/Documents/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[WARNING: Ignoring unknown variable 'filamentClipEnabled' in file gridfinity_baseplate.scad, line 194](https://github.com/ostat/gridfinity_extended_openscad/issues/194,/Users/belga/Documents/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[WARNING: Ignoring unknown variable 'filamentClipDiameter' in file gridfinity_baseplate.scad, line 195](https://github.com/ostat/gridfinity_extended_openscad/issues/195,/Users/belga/Documents/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[WARNING: Ignoring unknown variable 'filamentClipLength' in file gridfinity_baseplate.scad, line 196](https://github.com/ostat/gridfinity_extended_openscad/issues/196,/Users/belga/Documents/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[WARNING: Ignoring unknown variable '$idx' in file gridfinity_baseplate.scad, line 458](https://github.com/ostat/gridfinity_extended_openscad/issues/458,/Users/belga/Documents/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[WARNING: Ignoring unknown variable '$idx' in file gridfinity_baseplate.scad, line 458](https://github.com/ostat/gridfinity_extended_openscad/issues/458,/Users/belga/Documents/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[WARNING: Ignoring unknown variable '$idx' in file gridfinity_baseplate.scad, line 460](https://github.com/ostat/gridfinity_extended_openscad/issues/460,/Users/belga/Documents/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[WARNING: Ignoring unknown variable '$idx' in file gridfinity_baseplate.scad, line 461](https://github.com/ostat/gridfinity_extended_openscad/issues/461,/Users/belga/Documents/gridfinity_extended_openscad/gridfinity_baseplate.scad)
`

elbelga, 2024-02-01:
I believe is a problem with OpenScad version, with snapshot version works well

ostat, 2024-02-03:
Thanks for letting me know. 
I think I can add a check for the version, so the issue is clearer.

ostat, 2024-02-23:
I have added a check to assert version of OpenSCAD

closed 2024-02-23 by ostat

lied, 2024-04-30:
can you please put a info to the repo readme that the version of OpenSCAD needs to be the snapshot version? I've had the same issue (Mac OSX).