ostat/gridfinity_extended_openscad #10: [BUG] Newly added drawers do not have baseplates, even if they are enabled.

Monniasza, 2024-02-17:
When I enable baseplates on the drawer, the drawer does not have the bottom base, but throws a warning: `Ignoring unknown module 'grid_block'`. Please revert the base STL and my work workaround with tiling the base model. Also please merge the efficient shelves that I've made after you merged the PR.

ostat, 2024-02-17:
This should be fixed. Can you get latest and let me know if you are still having issues?
I also added some filament saving options to the walls. Feedback welcome.
I would like to avoid requiring the STL unless we absolutely need to. As it reduces the options, like setting the base to half pitch.
[image]

Monniasza, 2024-02-17:
It's okay. I'll check out the options. I'm using OpenSCAD version 2023.08.29 snapshot.

Monniasza, 2024-02-17:
The pattern extends past the front and starts too far to the front.
[image]

Monniasza, 2024-02-17:
Here are my settings:
[gridfinity_drawers.json](https://github.com/ostat/gridfinity_extended_openscad/files/14319404/gridfinity_drawers.json)

Monniasza, 2024-02-17:
On the left side of the drawer on the rear there should not be hex pattern.
[image]

ostat, 2024-02-17:
I will take a look. I had shift the side cutout to stop it going out the back. 

Good spot re the back will verify the math. 

Not sure if you saw, I added an option to the drawers to remove the floor and just have the grid. If you fill it with bins you might not need the solid floor. Need to check if the bins would stick through and mess up the sliding action.

ostat, 2024-02-17:
The wall coutout positioning is fixed in d64e655

Monniasza, 2024-02-18:
@ostat The wall cutout positioning looks good.
[image]

Monniasza, 2024-02-18:
However, the model with hex pattern takes forever to slice on PHOEU computer.

ostat, 2024-02-19:
I updated the model to allow for custom drawer sizer. It was more complex than expected, but I think I got there in the end. Would appreciate if you could review the changes in case I broke anything.
[image]

Monniasza, 2024-02-19:
I get lots of warnings and the drawer boxes do not appear, only the handles.
[image]
https://gist.github.com/Monniasza/8cce959ca7d661030b4a5329b9eec068

Monniasza, 2024-02-19:
Additionally, the drawerslides should only extend so far as the user has defined, but they always cover the entire width of the chest.

Monniasza, 2024-02-19:
You can define the inner dimensions of the drawers as a list.

ostat, 2024-02-19:
Sorry had not pushed all the changes. Think I got them all now.

Monniasza, 2024-02-19:
@ostat You have not said that setting drawer handle height to -1 makes it span the entire drawer height.

Monniasza, 2024-02-19:
Vertical handles do not require supports, unlike horizontal handles.

ostat, 2024-02-19:
We can prob close this now? I think all the issues are resolved. 
 
I have a couple of other ideas. 

I would like an option to swap drawers around when they are not the same size. Perhaps a small set (every 7mm) of runners that slides in to the walls of the drawer or chest? Though I'm a bit worried that the walls of the chest will flex too much. 

Also I would like an option to have the chest fit in to the shape of a Gridfinity bin. But maybe this would need to be a different script, as it would confuse the settings.

closed 2024-02-20 by Monniasza