ostat/gridfinity_extended_openscad #9: [BUG] Containers with labels don't stack properly

Monniasza, 2024-02-17:
If you stack another container on top of a container with a label, there is a gap between two containers, but there's no gap if a container is stacked on top of a spacer.
Left: container on top of a labeled container.
Right: container on top of a spacer.
[image]

ostat, 2024-02-17:
interesting, I think the fix would be to drop the label down a little. I can make it configurable. Do you have thoughts on what you think the default should be? 1mm?

ostat, 2024-02-17:
Checking the bins I have printed, If I stack a height 2 on a height 3, it matches the a height 5 bin I have.
One interesting thing is the top label is the same height as the divider, so if we need clearance on the label we also need it on the divider.
I wonder if the issue is without the label or divider, the tolerance in in the lip is allowing the bit to sit too low.

Monniasza, 2024-02-17:
@ostat The top of label should be about 0.2 mm below the bottom of the next bin when stacked.

ostat, 2024-02-18:
Are you thinking just a relief cut in for the label (some labels are quite thick). Or dropping both the top of the label and dividers (as they are both on the same z plane).

Monniasza, 2024-02-19:
I want to have adjustable height of the dividers.

ostat, 2024-02-19:
I have added two new options, 
- **zClearance**: drops the top of the label and top of dividers. Default is 0.
- **label_relief**: add relief to the label shelf to stick a label.
[image]

Monniasza, 2024-02-19:
The zClearance does nothing.

closed 2024-02-20 by ostat