ostat/gridfinity_extended_openscad #14: Feature Request: wall cutout location

benlangfield, 2024-04-23:
Would love the ability to move the location of the wall cutouts.  Currently they default to the centre of the bin and can't be moved.  I would like to be able to specify the location, for example, 2 grids or perhaps 80mm from the end of the x-axis.

This would be very handy for cutlery bins for kitchen drawers.  The wall gaps need to be lined up regardless of bin size for these as they are all sitting side by side, making it easier to get your finger in to grab a spoon sort of thing.

ostat, 2024-04-23:
I have changed the logic for the cutout option in changeset d5a8a1e.

The walls option used to turn on each wall at the center position. 
No we can use those value to change the position:

- 0: disables the cutout on that wall.
- Positive numbers: sets the position in Gridfinity units (42mm). 
  - I am considering supporting both GF units and mm. If the number is greater than the length of the wall in GF units, assume its mm.
- Negative numbers: position it using a ratio of `length/abs(value)`. I.E. -2 results in 1/2.

Its not that well tested, but I think it should do what we want.

[image]

benlangfield, 2024-04-23:
This is exactly what I wanted, thank you! 

[image]

Couldn't work out how to change the width, but wallcutout width of 0 seems about right anyway.

closed 2024-04-23 by benlangfield

ostat, 2024-04-23:
setting the `wallcutout width` to a number other than 0 will make the cutout the desired width. using 0 will use the `walllength/2`.
Is that not working for you?