ostat/gridfinity_extended_openscad #16: Odd Version message

galluccib, 2024-04-26:
Pulled the zip file down unzipped and opened "gridfinity_drawers.scad" and tried a few other and did the same thing.

I get the following error message - > ERROR: Assertion '(version()[0] > 2022)' failed: "This script requires a newer version of openSCAD. http://openscad.org" in file modules/gridfinity_modules.scad, line 217

I checked the website and it's OpenSCAD version 2021.01 which is the version I have installed... Any ideas what would be causing this?

Thanks,

galluccib, 2024-04-26:
Pulled down the nightly version and it seems to be working now...

closed 2024-04-26 by galluccib

ostat, 2024-04-30:
For any one getting the same error.
The release version of OpenSCAD is pretty old (its from 2021). This script requires features that are only in the development version. I will update the error message to make that clearer.

You can download that here - https://openscad.org/downloads.html#snapshots