ostat/gridfinity_extended_openscad #20: Support click lables

ostat, 2024-05-02:
Add support for - Gridfinity Swappable Labels by [Cullen J Webb](https://makerworld.com/en/@CullenJWebb)

https://makerworld.com/en/models/446624#profileId-353132

ostat, 2024-05-02:
[image]
[gridfinity_cup_modules_with_click_label.zip](https://github.com/ostat/gridfinity_extended_openscad/files/15187695/gridfinity_cup_modules_with_click_label.zip)

CullenJWebb, 2024-05-02:
I printed the test bin, thanks so much for adding this.

The labels fit well enough that I would consider it a success, but the dimensions are off by enough that the all important "click" isn't as satisfying. I've added the fusion360 files for both the bins and the labels, but here are the dimensions for reference:

-  Rib z height: 0.6mm
-  Rib X width at top: 1.3mm
-  Rib X width at bottom: 0.8mm

ostat, 2024-05-02:
I will try to adjust the ribs. I did them in a lazy way that is not very accurate. I will fix that up with more exact dimensions.

zerocle, 2024-05-31:
I commented on MakerWorld, but figure this is probably the better place. First, these bins and labels are amazing, so thank you for adding support for the click labels. There are a couple of bugs I noticed,

1. if the bin's split line intersects with the snap label, then the split will raise above the top of the label, interfering with the label placement
2. if you extend the label to take up multiple widths, only the first width of label gets the snaps embedded.

For reference my settings that I changed were
- width: 5
- vertical chambers: 5
- label style: click
- label width: 3

ostat, 2024-06-02:
@zerocle, Thanks for reporting. That is correct, there are a few complexities I need to resolve with the click labels.

The dividers is a bit tricky due to how I have written the code. (I would need to subtract the label from the divider, then subtract the label with click option from the bin space). One easy option might be to add a divider height option. This will allow the divider to be to be reduced below the click label height so it does not interfere. This might also be helpful for other situations.

Adding the click to other labels is something I have been meaning to do. I need to work out what to do if the label is not big enough to fit a standard click label. Perhaps @CullenJWebb has some ideas there?

CullenJWebb, 2024-06-03:
@ostat - The click labels standard will always be the same size, however, I do think I have a solution for undersized space for labels.

[image]

Here you can see that I've generated a bin with a label space of only 0.6u wide, whereas the click labels require at least 0.97u or something like that. I've added the click label slot negative volume and attached it over-hanging the edge.

When sliced it looks like this:

[image]

There are 2 ribs remaining to click into, with space for the label to overhang the edge. This gives users the option to manually trim their labels in the slicer or with scissors irl, or even let them hang over the edge. The over-hang isn't required though and if it's easier to you to simply trim off the right edge that would be fine as well.

So here are my suggestions:

- Allow for under-sized label slots, preferably by leaving an overhang for easier trimming
- Default to left-aligned slots and trim from the right side. All pre-generated labels place the size text on the left which is most likely to be the purpose of multiple smaller labels in a bin (multiple sized components).
- In the case of right aligned labels being selected then of course trim from the left side, the user will need to generate their own labels with right-aligned text (for now, if multiple people need something like this I would write a script to generate these as well)

**In short, I don't want to define a specific size for "mini" labels as we will always run into unique situations where somebody would prefer a larger or smaller label, and simply allowing them to trim to fit resolves all use cases for less than 1u width.**

Also, as I mentioned to zerocle on Makerworld, I like the idea of multiple label slots per bin, especially full width label slots. I've attached screenshots here showing how I added 3 label slots to a 3u wide bin using the negative volumes again.

[image]

That is the negative volumes being positioned, here is the sliced model:

[image]

ostat, 2024-06-04:
@CullenJWebb Some good ideas there. I will have a look and see what I can come up with. I think we can make some of that work.

One thought, The click labels hold in place using the triangle ribs. Did you consider instead integrating the click mechanism in to the long sides of the label? This should allow for a very similar click function without the need for the rib. Making the creating short and long label slots a lot easier.
For a bin that was say three units wide, you could just have a single long slot along the label and someone could insert the click labels are they want.

CullenJWebb, 2024-06-04:
@ostat Dang, that's a great idea. I will do some testing to see how well that works. I don't want to deprecate all of the bins people have already created so I need to be sure that this would be a worthwhile improvement with no downsides. I'll follow up with you on that.

CullenJWebb, 2024-07-05:
I am discussing the change of the click label standard with [ndevenish of gflabel](https://github.com/ndevenish/gflabel/issues/7). Feel free to add any thoughts you may have in that discussion, otherwise thank you for your patience while I redesign it.

Also, in case you are still working on implementing it, the name is going to be changed shortly to "Cullenect" labels to help differentiate the click labels from the click baseplates and lids already in the gridfinity ecosystem.

CullenJWebb, 2024-10-21:
@ostat The 2.0 version of the labels are ready here: https://github.com/CullenJWebb/Cullenect-Labels

They are being renamed "Cullenect" to differentiate between other swappable label types. Here are some of the features:

- Backward compatible with v1 bins (when generating 1u width labels)
- Variable label width
- Simpler geometry, easier to print and model.
- Emboss or deboss text
- More satisfying click
- Labels smaller than the socket can slide left and right along the new rail socket design
- Better vertical slot

And best of all it's been designed in OpenSCAD so that basic labels can be generated by users. Icons are still difficult in OpenSCAD though so I'm going to help get this implemented in gflabel as well.

This is my first project in OpenSCAD so there are likely some rough edges. You may be able to import the cullenect_socket() and cullenect_socket_negative() functions to implement them into the bins, but you are no doubt much better at this than me and the geometry is simple enough with this new version.

ostat, 2024-10-21:
@CullenJWebb, awesome I will take a look as soon as I have time.

ostat, 2024-10-26:
Support has been added for version 2. 
I need to improve the alignment logic (for all the label sockets) as the first cell has different alignment needs than the second and so on.
For Cullenect V2, I think the most compatible approach is to have the socket be the full width of the label/bin. 
@CullenJWebb, feedback welcome.

6072379

klingac, 2024-12-01:
Hi @ostat. I wanted to use those new Cullenect V2 labels, but it seems that they are not working with default options. See example in the picture
[image]
It looks like the label is "leaking" to bin's walls.

CullenJWebb, 2024-12-03:
@klingac I recommend using the negative volume provided by the Cullenect scad while this is being implemented. Once you've created the label you want you can also generate a negative volume and in your slicer position that where the label will go and you can customize any bin this way.

@ostat I think that full-width being the default makes sense, a lot of people will prefer that as 1U size labels will be able to fit in larger label sockets if users need to move their hardware to larger bins (blank labels could be used as spacers then). However, having the option for a 1U wide label would be terrific for bins where you don't want the label taking up space inside the cup

I downloaded the repo a few minutes ago an played around with the label settings. I'm sure that you're still making changes and tweaking things but I wanted to point out that the label size appears to be de-coupled from the width of the standard label support being generated for bins. 

Changing the labelSize value for X to 0.9 highlights the difference. Measuring in OpenSCAD it is about 2.95mm too large, resulting in the cut shown in the screenshot provided by @klingac when the size is left to the default.

[image]

ostat, 2024-12-06:
Labels are a bit of a mess at the moment.  Trying to make it possible to support all options with out glitching.

I am not sure how to get the width of the label to be correct for all scenarios.
I have also not updated to use your latest changes after the first release. I might have a chat to you about some tweaks to the function. I think I had to make some changes to bring it in. It would be handy if we could copy the function with no changes so its easier to keep it up-to date with any changes you make going FW.

CullenJWebb, 2024-12-15:
I'm working on the script still to add vertical socket generation (not relevant to your library) and while I'm in there I'll work on making the function more independent. I'll make is so that you pass the dimensions to it directly rather than a global variable.

ostat, 2025-01-24:
I have pushed a fix to better support V1 and V2.

@CullenJWebb, Feedback is welcome.

I struggled to make the standard sizes fit. Ideally the default size should be about 34mm, 36mm makes it a very 
**V2**
[image]

**V1**
[image]

CullenJWebb, 2025-01-29:
This works very well. I realize that the labels are a tight fit and I apologize for the trouble that's caused. The original spec was based on the bins generated with the Fusion360 plugin and the stackable rims on those are very different from vanilla Gridfinity. I would consider changing the spec but that would make V1 bins incompatible with V2 labels, not to mention everyone who has already printed V2 bins.

I've created a pull request here to auto calculate the width of the Cullenect sockets to spec unless the user specifies a size: https://github.com/ostat/gridfinity_extended_openscad/pull/115

I really like the solution you came up with to extrude above the label for clearance. It's not visible in the models that I printed and it looks very clean.

Labels snap in satisfyingly and I can configure many different bin/label configurations without any apparent bugs. I think that this is a solid and complete implementation.

ostat, 2025-01-29:
The print looks good. Thanks for testing. 

I think the change is fine, will accept later. 

I do have a concern with the widths. People need a way to create the right sized label for the bin/label size. 
This gets a bit tricky. 
If the bin is 3 units wide, split in two chambers with labels what size label should they make?
I can output the size to the console on OpenSCAD,
But that won't help people that use web generators or download pre generated models. 

If the bin/label side is 1unit then we can use a fixed set size to match the specification. 
With version 2 this is less of an issue. Im not sure the best solution.

CullenJWebb, 2025-01-29:
A single full width label socket is the solution. I've actually already printed test pieces and it works very well, I'll upload pictures later. 

It's oddly satisfying to slide the labels left and right inside the socket before adding other labels. Not useful, but satisfying.

CullenJWebb, 2025-01-29:
People can generate labels of any width they want and can throw the spec out the window. But combining labels of multiple units, even mixed units, works very well. A 6mm spacer is optional but it fills the gap created by the spec. 

Combining 2 labels will have 1 gap to fill, combining 3 labels will have 2, and so on.

Here's what that looks like. I printed fitment tests for a 3u and 2u wide socket:  

[image]

bmatheny, 2025-02-07:
I know there's still some work to be done, but I appreciate you all volunteering your time and working on this. I had been using the old V1 labels for a while (via gflabel) and the V2 changes address a lot of the annoyances I had. I'm looking forward to this getting finalized in extended, so I can start generating new bins with the right set of labels.

Anyhow, thank you both for putting your time in on this.

ostat, 2025-02-19:
@CullenJWebb and others,
Are we happy with the changes? Do we think we can close this now?
Of cause we can always, create new tickets if we find issues.

CullenJWebb, 2025-02-25:
@ostat I believe so, it's been working well for me and I agree that any future bugs should have their own ticket.