ostat/gridfinity_extended_openscad #21: Drawers: chest needs more clearance for drawer to fit properly

benlangfield, 2024-05-06:
I've printed a set of the drawers and the chest is an exact fit to the drawer so the drawer doesn't really fit in unless force.  There needs to be a bit of clearance added to the chest so the drawer slides in and out nicely.

ostat, 2024-05-15:
Thanks for rasising this. Sorry it didn't work out.
The Gridfinity bins are all undersized by 0.5mm, so a 1x is 41.5x41.5, a 2x is 83.5x83.5. The drawers currently have no clearance.
It the walls that are the issue, or the corners?

Happy to put in A clearance value, however that will change the drawer size, and the chest size.

ostat, 2024-05-15:
We do have `drawer_clearance` option. however its a bit broken and need to be reworked.

benlangfield, 2024-05-17:
I went back to it after it was sitting on the shelf for a while and it seems to work perfectly now.  Maybe the chest was still hot when I tried before and it needed to cool first?

If you add a clearance value I'd be concerned about drawer size changing as that means gridfinity stuff may not fit, I think it should only be chest size that changes.

ostat, 2024-05-18:
This was fixed in eb0d7e5

You can now configure both the clearance of the drawer and the chest.
If you add clearance the gridfinity items should still fit inside, but the drawer and the chest will increase in size

closed 2024-05-18 by ostat