ostat/gridfinity_extended_openscad #22: gridfinity_item_holder.scad - Add back label support

jprisant, 2024-05-07:
Reference: https://github.com/ostat/gridfinity_extended_openscad/pull/12

This release removed support for labels. Please add them back. Fwiw, the files are so wildly and wonderfully flexible, might as well keep all the options [instead of removing them] for folks to use [or not] as desired.

```openscad
abel = "disabled"; // ["disabled", "left", "right", "center", "leftchamber", "rightchamber", "centerchamber"]
// Width of the label in number of units, or zero means full width
label_width = 0;  // .01
```

vaz-ar, 2024-07-10:
@jprisant I was looking into this issue to see if I could fix it, but from what I am seeing the label has been put back in the file already, could you confirm ?

ostat, 2024-07-11:
@vaz-ar I think I have fixed it, but I have not really tested it. If you can confirm its working I am happy to close this one.

vaz-ar, 2024-07-11:
Hi @ostat, from what I am seeing it looks the same as it did before #12 was merged (I went back to the commit before that), the only difference is that the label is now in a dedicated section in the openscad customizer.

ostat, 2024-07-11:
I moved the label to its own section as it has more settings now. 
I think I will close this. Thanks for your help.

closed 2024-07-11 by ostat