ostat/gridfinity_extended_openscad #23: Wall Pattern Broken?

jprisant, 2024-05-07:
Is the wall pattern functionality broken?

When I try to use wall pattern in any of the primary .scad files, it only seems to work for one wall?

[image]

ostat, 2024-05-08:
I am not able to reproduce locally. I can see in the image you shared there are a bunch of errors though.

Would you mind double checking you have the latest, just in case its already fixed.
[image]

bbrendon, 2024-05-17:
I tried this on the web version and couldn't get the wallpattern working on more than one side. 

https://seasick.github.io/openscad-web-gui/?https%3A%2F%2Fraw.githubusercontent.com%2Fostat%2Fgridfinity_extended_openscad%2Fmain%2Fcombined%2Fgridfinity_basic_cup.scad

bbrendon, 2024-05-17:
It worked after I cloned the repo and opened the cup file. 
There should be some instructions on how to use this. It took me awhile to figure out you have to download everything and not just the cup file.

ostat, 2024-05-18:
You are right the online version is broken, I will get that fixed. Thanks for letting me know.
I will also update the doco to make it easier to get up and running.

ostat, 2024-06-10:
There is some strange caching happening with the seasick openscad-web-gui.

You can use the online version on Maker World which is now supported
https://makerworld.com/en/models/481168

jprisant, 2024-06-26:
> I am not able to reproduce locally. I can see in the image you shared there are a bunch of errors though.
> 
> Would you mind double checking you have the latest, just in case its already fixed.

I just saw this response! Sorry about the ridiculously late response. 

I redownloaded the repo today... I opened in OpenSCAD Dev version 2024.04.29 (git 97184930d) ... but still have the problem with the wall pattern... and now vertical separators don't seem to work either. 

[image]

I also extracted to a separate folder to open in OpenSCAD stable and experienced different issues.

I assume the problem is on my end, but I have no idea what else to try.

jprisant, 2024-07-26:
I just downloaded the newest openscad nightly and a copy of the gridfinity_extended repo, but I am still experiencing all these errors when trying to generate wall patterns
[image]

When I try to use the makerworld builder, the divider pattern  option doesn't work
[image]

ostat, 2024-07-26:
> I just downloaded the newest openscad nightly and a copy of the gridfinity_extended repo, but I am still experiencing all these errors when trying to generate wall patterns
> 
> [image]
> 
> 
> 
> When I try to use the makerworld builder, the divider pattern  option doesn't work
> 
> [image]
> 
> 

I have tried to reproduce the issue but have been unsuccessful. 
Could you please try download the repo as a zip. Unzip it to a new folder and run it from there? See how it goes.

ostat, 2024-07-26:
The divider pattern seems to only work on the non custom walls. 
Use the `Vertical chambers` option not the `Vertical chamber config`. I will look in to why that is not supported.

[image]

ostat, 2024-07-27:
The wall pattern divider walls now works on the custom configured walls. I have confirmed it works both locally and on makerworld.

[image]

[image]

ostat, 2024-08-06:
closed as fixed, @jprisant let me know if you still have issues we can re open

closed 2024-08-06 by ostat