ostat/gridfinity_extended_openscad #26: Extension tabs incorrectly positioned

jpypi, 2024-05-09:
Using `gridfinity_basic_cup.scad` and setting `position` to either "center" or "zero" (as opposed to the default: "default") and then with `extention_tabs_enabled = true` and either `extention_x_enabled = true` or `extention_y_enabled = true`, the extension tabs end up getting placed as separate pieces from the main bin.

This is just one example, but the different values of position and extension (x or y) put the tabs is various different places.
[image]

ostat, 2024-05-15:
Hi thanks for reporting this. I think this was a bug while I was trying to rework the tabs.
I believe it was fixed in 9e79f20. Would you be able to get latest and retry?

[image]

jpypi, 2024-05-18:
Pulled latest... still the same issue.

ostat, 2024-07-11:
Hi, Sorry I misunderstood this issue. Its now fixed. Thanks for raising

closed 2024-07-11 by ostat