ostat/gridfinity_extended_openscad #28: Allow non full sizes basplates to fill the sapce in a drawer

ostat, 2024-06-09:
Drawers will often not fit an exact gridfinity 42 unit size.
allow some options to allow the extra space to be filled.

ostat, 2024-06-09:
Updated the width and depth options to allow entering the size in GF units or in mm.
[image]

Added option to all the incomplete grid to filled, or cropped.

[image]
[image]

Monniasza, 2024-06-09:
This issue is misspelled.

ostat, 2024-06-10:
fixed in 8b19740

closed 2024-06-10 by ostat