ostat/gridfinity_extended_openscad #31: Magnet stress and removal groove

BreakMate, 2024-06-30:
I just want to start by saying, WOW! This resource is amazing for making custom gridfinity bins.

I wanted to request making a small addition to the magnet holes. If there is a small enough groove, this would allow for easier insertion and removal of the magnets. This can be applied to both the bins as well as the base, although I havent tested it with the base. I would add them myself, but I am horrid with openscad, but I was able to test this in Freecad with great success. Could you add a parameter to implement this?
[image]
[image]

ostat, 2024-06-30:
This is an interesting idea. I'm sure it can be added. 
Do you think we could angle it in to the corner?
My reason is the efficient bases. Those support magnets and have a cap that goes over the magnet. If we angle it in to the corner we won't need to change the cap.

BreakMate, 2024-07-01:
[image]

Good point! And quite a good feature, the efficient floor. I redrew the gaps and it still works! I made the straight part 1.5mm and the diameter of the groove 1.3mm. The magnets are 6mm and I made the holes in my model 6mm

BreakMate, 2024-07-01:
Using Prusaslicer to slice, MK4 with a 0.4 nozzle to print.

My initial reason for avoiding the outer corners was to reduce the chance of curling on the edges

ostat, 2024-07-01:
That looks good. Nice work getting it printed so fast. 
I agree that pointing to the centre would be more stable. 
I could direction it based on the selected floor type, efficient floors to the corner and smaller, otherwise to the centre and a bit bigger. 
Do you have thoughts on the relief size for different sized magnets?

BreakMate, 2024-07-01:
I do notice in your parameters, there is a hard cutoff of how close the magnets can get to the side of the base. not sure what that value is, but could be worked around.
If the magnets get smaller, then maybe it would hold better if the groove shrinks slightly. Making it too small would make it more difficult to fish it out (in the picture you can see the pick I'm using). Shouldn't be too big of an issue

If the efficient floor would account for the groove towards the center, I think that might work out the best and reduce the chance of curling.

BreakMate, 2024-07-01:
[image]
[image]

I love the efficient floor still able to use magnets, but I cant get them out now LOL

ostat, 2024-07-01:
I had a quick go at this tonight. 
I am bit sick so it needs a bit of work to get it working better and covering different sized magnets better. Also I have not tired to print this.

The code is pushed to the main branch.

[image]
[image]

BreakMate, 2024-07-01:
[image]
It works swell! 
I printed this with PETG in an enclosure. I've found that all gridfinity bins print better in an enclosure. It practically eliminates all warping. The black one I printed earlier I printed with PLA without an enclosure and the house fan was running, minor warping but enough to notice.

ostat, 2024-07-02:
Couple of questions;
Do you think this is good to close?

Based on your testing what are the recommend settings with the relief on and off?
The original design was to use 6.5mm diameter for a 6mm magnet, we add 0.1mm to the depth.
I would like to document recommendations for people.
Are the magnets sitting prod, in that image, do we need to recommend adding some extra depth for sagging of the bridge?

BreakMate, 2024-07-03:
I have never thought that 6.5mm was good for 6mm magnets (unless you mean for glue). When there are just magnet holes, 6.1 seems to do the trick. Thanks to this relief, I can make the magnet holes exactly 6 millimeters. It goes in with pressure, so I know that it won't come out easily.

As for the magnet protrusions, I believe that part of the problem comes from the efficient floor. The flex that is generated when trying to push in the magnets is causing them not to fully seat the first time around. Pressure must be applied from both sides. I haven't tested this with PLA, only PETG.

Magnet depth must be controlled by the user. I personally like 2.4 millimeters depth for a 2 millimeter magnet, probably due to overhangs. If overhangs are to be avoided, maybe add a version of the overhang remedy that only is meant to make bridges without necessarily making a hole.

Im in vegas currently, will be back thursday for more printed tests.

BreakMate, 2024-07-09:
Ok, so I found out that my magnets are actually 2.2 mm, so makes sense why 2.4mm height always worked well for me.

BreakMate, 2024-07-10:
[image]

It is kinda thin, maybe 1~2 more layers would be better

ostat, 2024-07-10:
That should be configurable as the floor thickness. Do you know what value was used? I would double check see if there are any bugs in the floor thickness and the clearance height calculations.

rwgunther, 2024-07-19:
I'm experiencing adhesion problems with the grooves pointing out to the corners. I have to use brims to print these without the corners curling. I don't seem to experience that as much when I don't include the grooves, but I definitely like having them!

ostat, 2024-07-27:
@rwgunther, I have change the `magnet release` option from a boolean (on/off) to a dropdown where you can select if you want the relief to point in or out. I think pointing in should resolve the adhesion issue you were having? Please let me know how it goes if you test it.
`Auto`, will point out for `efficient floor = on`, and in for `efficient floor = off`.

[image]

rwgunther, 2024-07-28:
This looks great. I'll try it out on my next print!

ostat, 2024-08-06:
Close as I think this is good now.

closed 2024-08-06 by ostat