ostat/gridfinity_extended_openscad #33: Butterfly clips (POC) impede cup seating on default base plates

rwgunther, 2024-07-28:
I realize you're still working on the base plate clips, so you're probably already aware of this. I've been experimenting with them, and I find that the butterfly clips stick out from the grid walls just enough that they keep the cups from sitting flat in the grid squares that use the clips. To get through a current project, I mixed up a tapered clip that does the job, but it's kind of a pain in the butt to use. You have to insert them with the correct side up, and they're harding to work with since they're a bit smaller. So I'm not sure I'd recommend this as a solution. Have you given this issue any thought yet? I'm also eager to see where you're going with the filament clip option.
[image]

ostat, 2025-01-23:
I have done some work on the connectors. 
I reduced the size of the butterfly connector.
I improved the positioning of the connectors (allowing options for mid wall or cell intersection).
I also added a new connector I called clip.

I need to do more testing but feedback is welcome

rwgunther, 2025-01-23:
Cool. I'm traveling now, but I'll test it out when I'm back home.