ostat/gridfinity_extended_openscad #35: please add fractional bases

ostat, 2024-08-08:
https://www.printables.com/model/630057-gridfinity-extended-openscad/comments/1975992

> Andrew Golovkov
> @Andor_1410728
> 
> Hey, what a great project.
> Could you please add fractional bases to complement fractional bins?
> Like these project but in openscad: https://www.printables.com/model/616289 and https://www.printables.com/model/641373-fractional-gridfinity-expanded

Andor, 2024-08-10:
Also, if I switch between "crop" and "fill" axis switches between length/width:
[image]
[image]

jziolkowski, 2024-11-21:
I was about to post the same request. Maybe it would be enough to "just" add "close" mode to oversize option, so that the last grid is properly closed, at whatever size it is? That will make a banger feature to use up these very last mm's of a drawer.

ostat, 2024-11-22:
This is now supported, I rewrote the base generator, so its more flexible.
I still need to do more testing.
[image]

jziolkowski, 2024-11-22:
OMG! I'll test later in the evening and will report back! Thanks!

jziolkowski, 2024-11-22:
I just cloned the repo. Also just installed openscad nightly.

[image]

So there is something still needed.

jziolkowski, 2024-11-22:
Oh wow, that's awesome! I was about to post that it still doesn't work, but it simply needs "fill" method. Amazing work, thanks A LOT!

ostat, 2024-11-23:
I should change the default.

jziolkowski, 2024-11-23:
Got another idea/request. 

In short, expanding of the "center grid functionality" so that I would be able to input some arbitrary size, that will be used to center the generated grid. Gridfinity rebuilt has this as "minimum grid size" and has the option to either center on that size, or to even align to either corner(s). Then, whatever the calculated grid size is (including the fractional grids), one would be able to additionally put the entire plate on an arbitrary size.

Practical example:

my drawer fits 7.5 x 9 grids, but there is still some free space left. There is no point in having a 7.586 fractional grid due to compatibility with the rest of the system, so that remaining "wasted" space could be filled with plastic so that it won't require glueing/screwing to the bottom of the drawer.

ostat, 2024-11-23:
So having the outer plate size be different from the grid size. I can take a look. I think you would need two options, the outer size and the alignment. 

Wouldn't it be just as effective to let the grid go to the end? You then have the option to fill the space with partial cups if you want. 
In your example the end cup could be x.586 units wide.

jziolkowski, 2024-11-23:
I would still prefer to have fractional grids/cups to .5 only, as that way it all maintains compatibility with everything (I really like the cup feature of having the bottom divided into 1/4ths)

Despite that drawer having dimensions I want to utilize to the max, having very odd sized grids would only work in the context of that drawer, and I want to have the cups interchangeable everywhere.

ostat, 2024-11-25:
This has been added
[image]

ostat, 2024-11-25:
You have the option to centre  or not the padding and center or not the grid. 

So if I'm not mistaken for the example you have shown, you would create a 3 x 3 with a larger outer size not centred, then create a 4 x 3 also not centred. 
I think this can do that example. Did you pull it down and give it a try?

jziolkowski, 2024-11-25:
Yes, I have the latest pull. I disabled the center option for the outer dimension (tho it doesn't seem to do much)

These are my options:
[image]

I need that padding added on top of the plate, not on the bottom.

ostat, 2024-11-25:
I see what you mean, I believe you can achieve that by switching your x and y. 

I am hesitant to increase the complexity where not needed. 
As much as I like the bit wise logic it's too complicated for people to use.

jziolkowski, 2024-11-25:
Ok, swapping X/Y does the trick. But maybe instead of the bitwise approach, you could change "center grid in outer <X> / <Y>" to "Outer <X> / <Y> alignment" with a dropdown "Left/Center/Right"

Then it could be aligned in every orientation and arguably easy to grasp to users

jziolkowski, 2024-11-25:
BTW did you consider other donation providers? Current paypal rates are borderline criminal.

jziolkowski, 2024-11-26:
This is awesome. Wow. Complete feature now :) Great job, thanks!

closed 2024-11-26 by ostat

jziolkowski, 2024-11-26:
> BTW did you consider other donation methods? Current paypal rates are borderline criminal.

@ostat ^

ostat, 2024-11-27:
@jziolkowski, thanks for the offer but its ok, you don't need to donate. Hope you find the scripts helpful.