ostat/gridfinity_extended_openscad #36: chest efficient back should cut per shelf

trysten, 2024-08-24:
[Efficient Back](https://github.com/ostat/gridfinity_extended_openscad/blob/8d511b6b2fc00955bb35ebd1b724416cffa3ea42/gridfinity_drawers.scad#L78) is a great idea. Considering that users will likely print drawer chests pointing upwards, efficient back should cut out per shelf, eliminating the bridges pictured. Thank you for this awesome project

[image]
[image]

ostat, 2024-08-24:
Oh, that looks like a bug. The back cutout looks rotated. Will take a look thanks.

ostat, 2024-09-14:
This has been fixed in commit 3681ebc

closed 2024-09-14 by ostat