ostat/gridfinity_extended_openscad #39: module_utility:92

wedgen, 2024-09-03:
Getting Parser Error: Syntax Error on this line.  I tried the demo/module_utility_test file and it also throws this error.  I will try to debug this later but wanted to start this maybe others also having issues.

wedgen, 2024-09-04:
I commented out `bentWall` and that let me compile.  I see that you require 2022+ which is a dev build.  Can you put a note in your README.md about this and what version you recommend?  Thanks!

ostat, 2024-09-04:
I can see there is a `Syntax Error` in the **module_utility.scad**. Some versions of OpenSCAD (makerworld for one), don't allow a trailing comma in the parameter list.
Its sloppy code, as the comma should not be there, but easy to miss as it does not cause an error in the windows version of OpenSCAD.

I will get it updated asap.

IzikMartinez, 2024-09-05:
One more parser error popped up after I adjusted the code on line 92 of module_utility.
While opening "gridfinity_item_holder.scad" after fixing like 92, a new parser error in line 104 of "module_item_holder" is thrown. 
The line is as follows. 
```
echo(str("🟩ItemGrid: count ", _hexGrid?hexGridCount:squareCount, " using grid ", _hexGrid?"hex":"square", ), input=hexGrid==true?"hex":hexGrid==false?"square":hexGrid, hexGridCount=hexGridCount,squareCount=squareCount);
```

I may have the implementation wrong, but I broke this up into 3 separate lines for easier debugging and readability and that seemed to solve the error on line 104. 

```
gridType = hexGrid ? "hex" : "square";

msg = str("🟩ItemGrid: count ", hexGrid ? hexGridCount : squareCount, " using grid ", gridType);

echo(msg);
```

However, now I've run into the same situation as wedgen where I see that you're using 2022+ for this project. On Linux, the latest easily available version is 2021.01, making this requirement something of an inconvenience.

Cheers.

ostat, 2024-09-15:
> One more parser error popped up after I altered the code on line 92 of module_utility to remove the trailing comma.
> 
> While opening "gridfinity_item_holder.scad" after removing the trailing comma on module_utility:92, a new parser error in line 104 of "module_item_holder" is thrown. The line is as follows.
> 
> ```
> echo(str("🟩ItemGrid: count ", _hexGrid?hexGridCount:squareCount, " using grid ", _hexGrid?"hex":"square", ), input=hexGrid==true?"hex":hexGrid==false?"square":hexGrid, hexGridCount=hexGridCount,squareCount=squareCount);
> ```
> 
> I may have the implementation wrong, but I broke this up into 3 separate lines for easier debugging and readability and that seemed to solve the error on line 104.
> 
> ```
> gridType = hexGrid ? "hex" : "square";
> 
> msg = str("🟩ItemGrid: count ", hexGrid ? hexGridCount : squareCount, " using grid ", gridType);
> 
> echo(msg);
> ```
> 
> Cheers.

This has been fixed, the issue was the trailing comma in the `str` function

ostat, 2024-09-15:
> see that you require 2022+ which is a dev build. Can you put a note in your README.md about this and what version you recommend? Thanks!

I have created a separate ticket, so I don't forget this

closed 2024-09-15 by ostat