ostat/gridfinity_extended_openscad #45: Improve the readme.md doco

ostat, 2024-09-15:
This file should include 
- a summary of the project.
- how to use it (download, clone or online)
- where the doco is
- what version of openscad you need

danielsan80, 2024-11-09:
@ostat could you give me in advance infos about Openscad version I need and which dev environment I need to contribute in developement (using git clone)?

For example I have some sintax errors like

```
 let(
    efficientFloor = validateEfficientFloor(settings[iCupBase_EfficientFloor]),
    magnetEasyRelease = validateMagnetEasyRelease(settings[iCupBase_MagnetEasyRelease], efficientFloor), // <------
  )
```

for the comma after the last var.

Maybe is a new feature of the `master` version of Openscad?

Thankyou

ostat, 2024-11-09:
Inside the developer version of OpenSCAD on windows. 
The trailing comma does not cause an issue on my setup. But it should not be there as it will cause issues with other setups, specifically I know makersworld does not like it. 
So I will get that fixed. 
I have been doing a bit of refactoring, to support some new features, I just have missed that one.

danielsan80, 2024-11-09:
The stable and downloadable version is 2021.01. The trailing comma is a good and practical feature but is not available in 2021.01 version. Anyway I'll try to switch to the dev version on Linux.

ostat, 2025-06-02:
done