ostat/gridfinity_extended_openscad #47: [Feature request]: Add rails to lids

jottr, 2024-09-30:
Thank you very much for sharing your awesome work. Here's a humble feature request.

[This lid design](https://www.printables.com/model/672541-gridfinity-1-x-2-size-inserts-can-be-closed-indivi) is very useful, since it 
1. allows for the sliding lids to be easily grabbed even when they are vertically flush
2. bins that are stacked on top don't slide off

@smkent already [implemented](https://github.com/smkent/monoscad/tree/main/gridfinity/rebuilt-bins) this lid design in openSCAD.  
His design is missing the very useful features of your bins like labels, finger scoops, etc.. 
It would be very useful if one could merge in this lid design. 

As a bonus, one could add an option to slide in plexiglass sheets - similar to what Matthew does with his [rugged box](https://www.printables.com/model/894202-modern-gridfinity-case/files) design.

ostat, 2024-11-08:
added fix in c6c4db4
[image]

For the slide in plexiglass, would that just be replacing the sliding lid with the right size piece of clear plastic?

jottr, 2024-11-08:
Will test this weekend. I'd be happy to provide feedback if desired. 
Thanks again.

ostat, 2024-11-08:
Thanks mate, Feedback is always welcome.

closed 2024-11-26 by ostat