ostat/gridfinity_extended_openscad #48: request: no lip, but full height

avxmw, 2024-10-03:
It does not currently seem possible, if it is pointers would be nice.

I'd like the option to remove the lip (already existing) *but* keep the height as if a lip was there. I want to be able to mix boxes with and without lips having a uniform height.

ostat, 2024-10-03:
I think what you are asking for is the minimal lip? 
The lips are 4.4mm that sit on top of the bin height. So a 3unit height bin is 21mm + 4.4mm lip. 

There are four lip styles. 
- **Normal**: 4.4mm largest rib. 
- **reduced**: 4.4mm smaller rib. 
- **minimal**: 4.4mm, same width as wall. 
- **none**: 0mm lip

https://docs.ostat.com/assets/openscad/gridfinity-extended/gridfinity_basic_cup-lip_style_text.gif

avxmw, 2024-10-03:
Thank you, I'll look into that. In case you're interested, Gridfinity Rebuilt (https://github.com/kennetek/gridfinity-rebuilt-openscad) does this as 1-click option which you might be interested to adopt.

ostat, 2024-10-03:
I am not sure what option you are referring to. Can you provide screenshots, with the option on and off describing what it does?

avxmw, 2024-10-04:
Surely, hopefully these screenshots help.

[image]

This shows where to find the option in 'gridfinity rebuilt'.

[image]

This hopefully illustrates what I'm interested in. 

GF-rebuilt allows to have the same total height of a box no matter if you want a stacking lip or not while GF-extended allows to cut off the lip, but loses height. 

There are times where the stacking is not needed, but actively in the way of reaching a tool or cleaning, for example in a tool drawer. Also for me it looks nicer if the height is uniform and I can just mix and match self-made and boxes from others.

I'm relatively sure this could be done with your version, but it is as of now not intuitive/noob friendly.

ostat, 2024-10-04:
Thanks for providing the extra info. Yes I think we can do something like that.

ostat, 2024-10-08:
I have extended the filled in option. We now have
 - `disabled`: not filled in
 - `enabled`: filled in
 - `filledinwithlip`: also fill in the lip.
 
[image]

Added in change 5f0fe56

closed 2024-10-08 by ostat

avxmw, 2024-10-09:
Thank you, that helps a lot :)

(now if openscad would someday gain export to .step I'd be happy)