ostat/gridfinity_extended_openscad #51: 1X1 Cup - Front missing

lsd87, 2024-10-07:
Using the basic cup (default design) and setting width X to 1.0 the front of the cup is missing, bit like having a wall cut out
Set the width to say 1.1 the front wall is correct.
Set the mm width to 42 the front wall is correct
Set the mm witch to 41.9 the front wall has the hole so I assume when set to 1 unit there's a rounding down so less than 42mm but not sure why
Using with Openscad 2024.09.30 

[image]

ostat, 2024-10-08:
This was a bug thanks for raising.
When the bin is smaller than 1 gf unit (42mm) the side lip is meant to be cut away to allow easier access to the bin.
A bug was introduced recently when I re-positioned the default location of the bin, resulting it it being off by 0.5 units (21mm)

fixed in 679f603

[image]

closed 2024-10-08 by ostat

lsd87, 2024-10-08:
Hi Chris

 

Thanks for coming back, noticed that when selecting X 1.0 on the new version then there’s 2 cut-outs on the sides, set the size to 42mm they disappear

 

Also

I can raise an issue (if you want J) about the click label, they don’t fit – I think the recess is back to far.

Found the attached scad file from somewhere that works but does not have all the great options your version has

 

Thanks

 

Leigh

 

image.png (view on web) <https://github.com/user-attachments/assets/e65ddbf0-5aee-40bd-8485-590ff9a75be2>

lsd87, 2024-10-08:
Bit of info - 
If I change from the default to 1.0 using the down arrow button, at 1.0 you get the side notches.
Change to 1.1 with the up arrow, notches get removed, set back to 1.0 they return
Enter 1 manually and then click up (1.1) and then down back to 1.0 the notches do not appear.
Go down to (down arrow) 0.9, notches appear, go back up to 1.0 using up arrow they disappear
SO once you have set 1.0 manually it SCAD appears to remember its 1.0 and not 0.999999....

ostat, 2024-10-08:
That sounds about right. The error is a common programming _bug_ related to floating point numbers. Pressing the up or down arrow many times the error accumulates. 
I could solve it in the script two ways. 
- Round the number to two decimal places. 
- apply the cutouts after 0.99 instead of 1. 

I run into this type of error quite a bit in OpenSCAD, I always just ignore it as it's easy fixed. But it might be worth putting in a fix for others who might not realise the cause.

lsd87, 2024-10-09:
Looks like option 2  0.99 might be the easiest option

BTW. I assume the cuts are for getting your fingers in when you have a narrow cup

ostat, 2024-10-10:
I added rounding to dimensions that should fix the issue.
4dc4eb