ostat/gridfinity_extended_openscad #53: Click Labels do not fit

lsd87, 2024-10-08:
When printed the click/snap labels do not fit into the recess, It looks like the recess goes back to far.
Adjusting the zclearance does not help.

The scad file below does work, I think its one of your older versions 17/2/2024
 
[gridfinity_basic_cup_simplified.zip](https://github.com/user-attachments/files/17292812/gridfinity_basic_cup_simplified.zip)

lsd87, 2024-10-08:
Bit of testing
If you set the lip style to reduced/minimum/not stackable then you can see the full recess, printed the not stackable version and the click/snap label fits fine. 
Set to reduced lip and zone in 
[image]
Then select normal and see the lip "eat" into the label platform at the back
[image]

lsd87, 2024-10-08:
A work around is to set the label depth to 16.5mm or more so that the curves of the back wall don't eat into the label recess.
Also setting zclearance to 1.0mm helps with stacking

lsd87, 2024-10-09:
I had a look at the old simplified SCAD cup that I printed and the whole label platform does not touch the inner edge lip because on the latest cups there's an extra lip I assume for better stacking.
So I printed another cup with the zclearance set to 3mm, this made the click label fit.
When label style is normal, the label relief can be used to lower the insert part of the platform but when click style is set, the relief setting does not work, that's why I had to use the zclearance

This is how the lip/platform connect with zclearance set to 0
[image]

And this is when its set to +3
[image]

The problem with labels is that there appears to be at least 4 types, I would like to use ndevenish great gflabel program so that I can make up any label I need without looking around the internet to see if someone else has created one using fusion/freecad ...etc
[https://github.com/ndevenish/gflabel#label-bases](url)

The pred labels don't appear to fit because the tabs need a slot in the cup edge to fit in (may be able to shrink using the --width option in gflabel), the predbox and plain might work but I liked the cullen/webb labels because they would be easy to change

I think its best to stick with the standard size (is there one) label initially. but with your version allows a user to adjust the width/size of the platform AND the recess. that way all variations could be covered.

As for click settings default 
First does the zclearance just alter the location of the label platform ? if so, it needs to be under the Label settings section
Second can you put a note to say "if you use click, set zclearance to say 3mm" or something like it or 
setting click sets the zclearance to +3 automatically

Sorry can't think of a good fix,  too many variables and I know nothing about SCAD

lsd87, 2024-10-09:
Sorry about the close/reopen, clicked on the wrong button hoh

lsd87, 2024-10-15:
I think the main problem is with the lip style
If the click recess is supposed to be 36.7mm, then using the normal lip, I read 36mm between the lip, therefore the click label will not fit. Same for the depth, 11.3 label won't fit in a 10mm recess. 
Its either not use the normal lip or move the label platform down below the curved lip about 2mm or more zclearance

Normal lip
[image]

Reduced lip
[image]

Normal lip plus 2mm
[image]

BTW. Assume you know about the Line 177 error
[WARNING: variable extension_enabled not specified as parameter in file gridfinity_basic_cup.scad, line 177](https://github.com/ostat/gridfinity_extended_openscad/issues/177,D:/lsd/Downloads/gridfinity_extended_openscad-main%2015th%20Oct/gridfinity_extended_openscad-main/gridfinity_basic_cup.scad)
[WARNING: variable extension_tabs_enabled not specified as parameter in file gridfinity_basic_cup.scad, line 177](https://github.com/ostat/gridfinity_extended_openscad/issues/177,D:/lsd/Downloads/gridfinity_extended_openscad-main%2015th%20Oct/gridfinity_extended_openscad-main/gridfinity_basic_cup.scad)
[WARNING: variable extension_tab_size not specified as parameter in file gridfinity_basic_cup.scad, line 177](https://github.com/ostat/gridfinity_extended_openscad/issues/177,D:/lsd/Downloads/gridfinity_extended_openscad-main%2015th%20Oct/gridfinity_extended_openscad-main/gridfinity_basic_cup.scad)

lsd87, 2024-10-15:
I think using the gflabel program may be around the click label sizes since width and depth can be set so click (webb) default sizes can be over ridden.

usage: gflabel [-h] [--base {pred,plain,none,webb,predbox}] [--vscode] [-w WIDTH]
               [--height HEIGHT] [--depth DEPTH_MM] [--no-overheight] [-d DIVISIONS]
               [--font FONT]
               [--font-size-maximum FONT_SIZE_MAXIMUM | --font-size FONT_SIZE]
               [--font-style {regular,bold,italic}] [--margin MARGIN]
               [-o [OUTPUT ...]] [--style {embossed,debossed,embedded}]
               [--list-fragments] [--list-symbols] [--label-gap LABEL_GAP]
               [--column-gap COLUMN_GAP] [-v]
               LABEL [LABEL ...]

For me the various options are to
1). Drop the platform down when using click labels
2). I Gflabel set the sizes manually
3). Just use std labels and stick then in using double sided tape

The pret labels are 35.5mm long with 2 tabs making them 37.5mm long, would be nice to use them as well, would need a recess in the lip to take the two tabs - but this is a feature update :-)
https://www.printables.com/model/592545-gridfinity-bin-with-printable-label-by-pred-parame

ostat, 2024-10-15:
> BTW. could you make the click label "eat" into the lip like the pret ones ?
I am sure we could. however it would require reworking how the label is implemented, and is not an easy fix.
The label is subtracted from the cavity, and then the cavity is subtracted from the filled in bin.
This means even if the label extends past the cavity it cant interfere with the wall or the lip of the bin. 
So we would need to find a clean way around this.

lsd87, 2024-10-17:
I've been playing with gflabel and the problem with the normal platform is that even lowering the platform down by say 3mm, the rear had the rounded edges at the corners. SO you have to resize the platform so the label to comes out away from the rounded corners, or cut the corners off the label
[image]

Using the click you have a square recess corners
[image]

Could you make the normal be like the click but without the 3 rails ?

CullenJWebb, 2024-10-21:
@ostat The 2.0 version of the labels are ready here: https://github.com/CullenJWebb/Cullenect-Labels

They are being renamed "Cullenect" to differentiate between other swappable label types. Here are some of the features:

- Backward compatible with v1 bins (when generating 1u width labels)
- Variable label width
- Simpler geometry, easier to print and model.
- Emboss or deboss text
- More satisfying click
- Labels smaller than the socket can slide left and right along the new rail socket design
- Better vertical slot

And best of all it's been designed in OpenSCAD so that basic labels can be generated by users. Icons are still difficult in OpenSCAD though so I'm going to help get this implemented in gflabel as well.

ostat, 2024-10-26:
I am going to close this ticket, as we have #20 still open that relates to adding support for the clicklabels

closed 2024-10-26 by ostat