ostat/gridfinity_extended_openscad #56: Wall cutouts margin behavior not consistent with subdivisions

shrestha-pranav, 2024-10-17:
When creating a basic cup with wall cutouts, the internal dividers don't adhere to the global crop/space settings. For instance, with this hexgridrotated pattern, you can see some the internal dividers are cropped off with no margins at all, even though the outside walls have margins.
[image]

<details>
<summary> Parameter Set </summary>

```json
"example_bin": {
    "box_corner_attachments_only": "false",
    "cavity_floor_radius": "-1",
    "center_magnet_diameter": "0",
    "center_magnet_thickness": "0",
    "chamber_wall_thickness": "1.2",
    "chamber_wall_zClearance": "0",
    "cutx": "0",
    "cuty": "0",
    "depth": "[3, 0]",
    "efficient_floor": "smooth",
    "enable_help": "false",
    "extension_tab_size": "[10, 0, 0, 0]",
    "extension_tabs_enabled": "true",
    "extension_x_enabled": "disabled",
    "extension_x_position": "0.5",
    "extension_y_enabled": "disabled",
    "extension_y_position": "0.5",
    "filled_in": "disabled",
    "fingerslide": "none",
    "fingerslide_radius": "8",
    "fingerslide_walls": "[1, 0, 0, 0]",
    "flat_base": "false",
    "floor_thickness": "0.69999999999999996",
    "half_pitch": "false",
    "height": "[14, 0]",
    "hole_overhang_remedy": "2",
    "horizontal_chambers": "4",
    "horizontal_irregular_subdivisions": "false",
    "horizontal_separator_bend_angle": "0",
    "horizontal_separator_bend_position": "0",
    "horizontal_separator_bend_separation": "0",
    "horizontal_separator_config": "10.5|21|42|50|60",
    "horizontal_separator_cut_depth": "0",
    "label_position": "left",
    "label_relief": "0",
    "label_size": "[0, 14, 0, 0.6]",
    "label_style": "disabled",
    "label_walls": "[0, 1, 0, 0]",
    "lip_style": "normal",
    "magnet_diameter": "6.2000000000000002",
    "magnet_easy_release": "off",
    "position": "zero",
    "screw_depth": "6",
    "sliding_clearance": "0.10000000000000001",
    "sliding_lid_enabled": "false",
    "sliding_lid_thickness": "0",
    "sliding_min_support": "0",
    "sliding_min_wallThickness": "0",
    "spacer": "false",
    "tapered_corner": "none",
    "tapered_corner_size": "10",
    "tapered_setback": "-1",
    "vertical_chambers": "1",
    "vertical_irregular_subdivisions": "false",
    "vertical_separator_bend_angle": "0",
    "vertical_separator_bend_position": "0",
    "vertical_separator_bend_separation": "0",
    "vertical_separator_config": "10.5|21|42|50|60",
    "vertical_separator_cut_depth": "0",
    "wall_thickness": "1.2",
    "wallcutout_horizontal": "disabled",
    "wallcutout_horizontal_angle": "70",
    "wallcutout_horizontal_corner_radius": "5",
    "wallcutout_horizontal_height": "0",
    "wallcutout_horizontal_position": "-2",
    "wallcutout_horizontal_width": "0",
    "wallcutout_vertical": "enabled",
    "wallcutout_vertical_angle": "70",
    "wallcutout_vertical_corner_radius": "5",
    "wallcutout_vertical_height": "0",
    "wallcutout_vertical_position": "-2",
    "wallcutout_vertical_width": "0",
    "wallpattern_dividers_enabled": "horizontal",
    "wallpattern_enabled": "true",
    "wallpattern_fill": "crop",
    "wallpattern_hole_sides": "6",
    "wallpattern_hole_size": "5",
    "wallpattern_hole_spacing": "1.5",
    "wallpattern_style": "hexgridrotated",
    "wallpattern_voronoi_noise": "0.75",
    "wallpattern_voronoi_radius": "0.5",
    "wallpattern_walls": "[1, 1, 1, 1]",
    "width": "[2, 0]",
    "zClearance": "0"
}
```

</details>

Similar but separate issue: If the grid is taller than a certain size (with default wall_thickness=0), the dividers have a flat layer on the back of the grid pattern layer. Setting wall thickness to 1.2 fixes this issue (see sample parameters above). I believe this is because the wall cutout patterns have a default depth that is less than 1.6

[image]
[image]

ostat, 2024-10-17:
Thanks for raising. I kinda new it would be an issue, but knew it would be complicated so ignored it.
I had a go at fixing this, I don't think its right yet but its a start 11fd42b

shrestha-pranav, 2024-10-20:
That was fast, wow. This solves my issue! I should've waited a few days before printing my bins hah

[image]

For issue #2, if you set your height to >= 14 units, you will see a back wall on the dividers. Setting wall_thickness=1.20 will force thinner dividers (from default 1.60 I believe) and the back wall disappears (like in the first picture above)

<details>
<summary>Parameter set example for back wall issue </summary>

```json
"Cutout 2x3": {
    "box_corner_attachments_only": "false",
    "cavity_floor_radius": "-1",
    "center_magnet_diameter": "0",
    "center_magnet_thickness": "0",
    "chamber_wall_thickness": "1.2",
    "chamber_wall_zClearance": "0",
    "cutx": "0",
    "cuty": "0",
    "depth": "[3, 0]",
    "efficient_floor": "smooth",
    "enable_help": "false",
    "extension_tab_size": "[10, 0, 0, 0]",
    "extension_tabs_enabled": "true",
    "extension_x_enabled": "disabled",
    "extension_x_position": "0.5",
    "extension_y_enabled": "disabled",
    "extension_y_position": "0.5",
    "filled_in": "disabled",
    "fingerslide": "none",
    "fingerslide_radius": "8",
    "fingerslide_walls": "[1, 0, 0, 0]",
    "flat_base": "false",
    "floor_thickness": "0.69999999999999996",
    "half_pitch": "false",
    "height": "[14, 0]",
    "hole_overhang_remedy": "2",
    "horizontal_chambers": "4",
    "horizontal_irregular_subdivisions": "false",
    "horizontal_separator_bend_angle": "0",
    "horizontal_separator_bend_position": "0",
    "horizontal_separator_bend_separation": "0",
    "horizontal_separator_config": "10.5|21|42|50|60",
    "horizontal_separator_cut_depth": "0",
    "label_position": "left",
    "label_relief": "0",
    "label_size": "[0, 14, 0, 0.6]",
    "label_style": "disabled",
    "label_walls": "[0, 1, 0, 0]",
    "lip_style": "normal",
    "magnet_diameter": "6.2000000000000002",
    "magnet_easy_release": "off",
    "position": "zero",
    "screw_depth": "6",
    "sliding_clearance": "0.10000000000000001",
    "sliding_lid_enabled": "false",
    "sliding_lid_thickness": "0",
    "sliding_min_support": "0",
    "sliding_min_wallThickness": "0",
    "spacer": "false",
    "tapered_corner": "none",
    "tapered_corner_size": "10",
    "tapered_setback": "-1",
    "vertical_chambers": "1",
    "vertical_irregular_subdivisions": "false",
    "vertical_separator_bend_angle": "0",
    "vertical_separator_bend_position": "0",
    "vertical_separator_bend_separation": "0",
    "vertical_separator_config": "10.5|21|42|50|60",
    "vertical_separator_cut_depth": "0",
    "wall_thickness": "1.2",
    "wallcutout_horizontal": "disabled",
    "wallcutout_horizontal_angle": "70",
    "wallcutout_horizontal_corner_radius": "5",
    "wallcutout_horizontal_height": "0",
    "wallcutout_horizontal_position": "-2",
    "wallcutout_horizontal_width": "0",
    "wallcutout_vertical": "enabled",
    "wallcutout_vertical_angle": "70",
    "wallcutout_vertical_corner_radius": "5",
    "wallcutout_vertical_height": "0",
    "wallcutout_vertical_position": "-2",
    "wallcutout_vertical_width": "0",
    "wallpattern_dividers_enabled": "horizontal",
    "wallpattern_enabled": "true",
    "wallpattern_fill": "crop",
    "wallpattern_hole_sides": "6",
    "wallpattern_hole_size": "5",
    "wallpattern_hole_spacing": "1.5",
    "wallpattern_style": "hexgridrotated",
    "wallpattern_voronoi_noise": "0.75",
    "wallpattern_voronoi_radius": "0.5",
    "wallpattern_walls": "[1, 1, 1, 1]",
    "width": "[2, 0]",
    "zClearance": "0"
}
```

</details>

Backwall pictures:
[image] [image]

ostat, 2024-10-26:
Thanks for posting the config.
I used your config, and believe this is fixed.
[image]

shrestha-pranav, 2024-11-07:
Looks great to me. Feel free to close the issue as fixed :)

Angle is good to me, I'd rather specify angle than have to calculate the lengths for this use case. In other use cases, length might make sense, although it's more of a convenience factor since they can be calculated either way.

ostat, 2024-11-08:
done

closed 2024-11-08 by ostat