ostat/gridfinity_extended_openscad #58: Pred labels produce errors

lsd87, 2024-11-09:
Get this lot when selecting pred labels
(Yep me again and my label fetish :-) )

[WARNING: undefined operation (vector + number) in file modules/module_gridfinity_label.scad, line 220](https://github.com/ostat/gridfinity_extended_openscad/issues/220,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)
[WARNING: max() parameter could not be converted: argument 1: expected number, found undefined (undef) in file modules/module_gridfinity_label.scad, line 220](https://github.com/ostat/gridfinity_extended_openscad/issues/220,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)
[WARNING: undefined operation (number + undefined) in file modules/module_gridfinity_label.scad, line 220](https://github.com/ostat/gridfinity_extended_openscad/issues/220,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)
[WARNING: undefined operation (number - vector) in file modules/module_gridfinity_label.scad, line 220](https://github.com/ostat/gridfinity_extended_openscad/issues/220,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)
[WARNING: undefined operation (undefined - number) in file modules/module_gridfinity_label.scad, line 220](https://github.com/ostat/gridfinity_extended_openscad/issues/220,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)
[WARNING: Unable to convert translate([0, undef, undef]) parameter to a vec3 or vec2 of numbers in file modules/module_gridfinity_label.scad, line 220](https://github.com/ostat/gridfinity_extended_openscad/issues/220,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)
[WARNING: undefined operation (vector + number) in file modules/module_gridfinity_label.scad, line 221](https://github.com/ostat/gridfinity_extended_openscad/issues/221,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)
[WARNING: Unable to convert cube(size=[168, 7, undef], ...) parameter to a number or a vec3 of numbers in file modules/module_gridfinity_label.scad, line 221](https://github.com/ostat/gridfinity_extended_openscad/issues/221,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)
[WARNING: undefined operation (vector + number) in file modules/module_gridfinity_label.scad, line 244](https://github.com/ostat/gridfinity_extended_openscad/issues/244,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)
[WARNING: max() parameter could not be converted: argument 1: expected number, found undefined (undef) in file modules/module_gridfinity_label.scad, line 244](https://github.com/ostat/gridfinity_extended_openscad/issues/244,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)
[WARNING: undefined operation (number + undefined) in file modules/module_gridfinity_label.scad, line 244](https://github.com/ostat/gridfinity_extended_openscad/issues/244,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)
[WARNING: undefined operation (number - vector) in file modules/module_gridfinity_label.scad, line 244](https://github.com/ostat/gridfinity_extended_openscad/issues/244,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)
[WARNING: undefined operation (undefined - number) in file modules/module_gridfinity_label.scad, line 244](https://github.com/ostat/gridfinity_extended_openscad/issues/244,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)
[WARNING: Unable to convert translate([0, undef, undef]) parameter to a vec3 or vec2 of numbers in file modules/module_gridfinity_label.scad, line 244](https://github.com/ostat/gridfinity_extended_openscad/issues/244,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)
[WARNING: undefined operation (vector + number) in file modules/module_gridfinity_label.scad, line 245](https://github.com/ostat/gridfinity_extended_openscad/issues/245,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)
[WARNING: Unable to convert cube(size=[82.8, 14, undef], ...) parameter to a number or a vec3 of numbers in file modules/module_gridfinity_label.scad, line 245](https://github.com/ostat/gridfinity_extended_openscad/issues/245,D:/lsd/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_label.scad)

MinchinWeb, 2024-12-06:
Related to #66?

ostat, 2025-01-24:
This issue has been fixed.
Feedback is welcome.

[image]

lsd87, 2025-01-25:
Yep that's fixed - this can be closed - Thanks for your help, looks like you've been busy recently :-)

closed 2025-01-25 by lsd87