ostat/gridfinity_extended_openscad #60: Subdivision separators float above floor using flat base

rwgunther, 2024-11-15:
When I create a basic cup with subdivisions, the separators don't go all the way down to the floor if you use the **flat base** option. I've found that I can work around this by enabling the **efficient floor** option, but I'm thinking this is probably a bug. I've tried it both in MakerWorld's customizer and locally in OpenSCAD.

[image]
[image]

ostat, 2024-11-24:
I believe this is now fixed, its not yet uploaded to makerworld, will do that soon.
[image]

rwgunther, 2024-11-24:
Excellent! I'll check it out. Thanks.

ostat, 2024-12-08:
believed to be fixed

closed 2024-12-08 by ostat