ostat/gridfinity_extended_openscad #61: Cup extensions not compatible with mm-based width/depth

cjlucas, 2024-11-17:
I'm having an issue printing a bin that's 8 GF units deep and 3.5 GF units wide (or 147mm). My printer bed isn't big enough to print something that deep, so I'm attempting to use the extension feature to print the cup in two section, one that's 5 GF units deep (4.5 after setting the extension to "front") and one that's 4 GF units deep (3.5 after setting the extension to "front"). Because I need to rotate one of the parts to join with the other, the half width GF unit ends up on the wrong side. 

Is there something I'm missing that would allow me to accomplish that? 

Thanks for your work on this, it's made building cups so much easier.

ostat, 2024-11-17:
I cant recall (not at my pc) but my intention is you would config the final bin (8 units), then you would pick where to split and render the front and then back of the split separately. 

This is different to the original design but should be easier to understand.

cjlucas, 2024-11-17:
Thanks for the quick reply!

That's how I understood it to work based on the official documentation, so I think it may just be a bug around the extension position variables. Regardless of whatever `extension_y_position` is set to, when `extension_y_enabled` is set to `front`, it'll always render the cup with a depth of n-0.5 GF units, and if it's set to `back`, it'll always render at 0.5 GF units.

ostat, 2024-11-17:
I tested today, you are right that is a problem. I will investigate ASAP

ostat, 2024-11-25:
This has been fixed.
[image]

cjlucas, 2024-11-25:
Thanks for making a fix for this! Could you double check that the `extension_y_position` is working as well? Adjusting the `x` position works as expected, but changing the `y` position doesn't change the model. It stays at its normal 0.5 cut.

ostat, 2024-11-25:
Y was using the X dimension, nice catch thanks.
[image]

cjlucas, 2024-11-25:
Thanks @ostat! I confirmed both the X and Y extensions are working as expected.

ostat, 2024-11-26:
thanks for confirming

closed 2024-11-26 by ostat