ostat/gridfinity_extended_openscad #72: Feature request: option to disable rounded outside corners for the baseplate

jziolkowski, 2024-11-23:
Since custom grid is not compatible with the newly improved fill mode for fractional grids, it'd be nice to either toggle all rounded outside edges of the plate, or apply the same 4-bit array to control which ones are round?

ostat, 2024-11-23:
I was not sure if anyone really used the custom grids. I liked the 4bit array, but it was kinda complex, I would like to make it auto calculate but have not tried to create that logic.

What kind of shape were you trying to create?

jziolkowski, 2024-11-23:
Like this, but with marked edges being straight, not rounded
[image]

ostat, 2024-11-23:
using the bitwise style option would not be easy for this. Also there are some issue with cropping I would need to deal with.

I have added an option to set the plate corner radius. I think this is what you want?
[image]

jziolkowski, 2024-11-23:
This is more than enough! Fantastic. Thanks!

closed 2024-11-23 by jziolkowski