ostat/gridfinity_extended_openscad #75: Subdivider height interferes with sliding lid

meneelyt, 2024-11-29:
When I add internal dividers to a cup with a sliding lid, the internal walls are too high and prevent the lid from being used.

I'm using standalone OpenScad, with a pull from the github repo on 2024-11-27.

Kudos on the sliding lid - on the entire project, of course, but last year I spent many hours trying to come up with a nice sliding lid design and ended up unsatisfied; yours if far nicer out of the box. Thanks.

Tim Meneely

ostat, 2024-11-29:
Hi, thanks for raising. The clearance is meant to be adjusted to account for the sliding lid, so that is def a bug.
You can do it manually by editing the `chamber_wall_zclearance`. Might help until I can take a look at the issue.

ostat, 2024-11-29:
This should be fixed now ef83117

[image]

closed 2024-11-29 by ostat