ostat/gridfinity_extended_openscad #78: Feature idea: grids connected on crossing

jziolkowski, 2024-12-02:
Basically what this guy did, but at normal gridfinity sizes

https://www.youtube.com/shorts/LLesH2ijoC0

jziolkowski, 2025-01-23:
yo this is awesome, i'll test later today when printing another drawer

does it also have the option to generate the clips?

ostat, 2025-01-23:
Yes, there is an option called, "**connectors only**"

[image]

ostat, 2025-01-23:
positioning the connectors on the cell side was simple, on the cell intersections was somewhat clunky. I position them pre cell, and cells share walls. I also dont want want clips on cells with padding or ones what are not full size.

then making it work for the printer bed splitting feature, for example

[image]

Anyway let me know your thoughts,

jziolkowski, 2025-02-04:
Ok that took a while, I was ill.

Anyway, these rock. Snapping is very tight, but when carefully using pliers or an edge of something hard it goes in very snugly. Printed with PLA and it didn't crack so that's nice, but I guess for mass production I'll go with PETG for the connectors.

If I'd be extra picky, if you could shave off like 0.5mm from the model on end, then it will give some breathing room for minor print inconsistencies.

Overall, A+++++

ostat, 2025-02-04:
Hi, thanks for testing the feature. 
How did you find the bins sit? The first couple of tries I felt the clip was a little raised.

When you say from the connectors end do you mean the length? Meaning if you select 10mm, have the slot be 10mm wide and the connector be 9.5mm long?

There is a tolerance parameter for the clips, but I think it needs work. the tolerance increases the clearance in all dimensions including the length between the base plate and the clip.  Currently it affects both base plate and the clip. Which I am not sure about. Say you print a base with a 0.1 tolerance, then print some clips and tweak the tolerance to dial in the fit. When you go back to print more bases it will also use the new tolerance and the fit will be different.
I think the base should prob be fixed dimensions, and the tolerance should only impact the clip. I did it this was as the parts are quite thin shaving 0.5 off the wall of the clip makes it quite small, where 0.25 off both the base and the clip is not so bad.

jziolkowski, 2025-02-04:
For now I did not test your bins, I tested a 2x2 shallow 18650 holder (it weighs quite a lot with the accus, so that helped). I did not observe anything odd with fitting - I think why would it? The connectors sit flush with the top line of the grid and the outer cross section has the same shape as the grid.

One thing I did observe was that when I joined 4 grids with the center piece, they were buckled a bit, pulling upwards from the surface. In practice this is only noticeable when not under load. I will test that again but I will slightly stretch the center cross along X/Y so the fit is not as snug.

> When you say from the connectors end do you mean the length? Meaning if you select 10mm, have the slot be 10mm wide and the connector be 9.5mm long?

yes, trim them lengthwise for the straight connectors, and along X/Y for T/X connectors.

At the same time, maybe I need to tune the printer a bit more and adjust the print settings. For example I've noticed that when printing the grid it's better to move the seam to be not in the corner - when filament accumulates there it lowers the fitting tolerance.

Still, the initial test is 99% successful IMO. Knowing that after few days the grid will conform to the weight due to PLA characteristics I'm sure it will be flat :)

jziolkowski, 2025-02-04:
oh wait, my bad. I did test your bin. 2x1 with some PCBs inside. It sat well, no binding at all, and the expected wiggle room.

Monniasza, 2025-02-09:
@ostat It works, but it's little tight for me. I'll post a make on Printables soon.

ostat, 2025-02-09:
> @ostat It works, but it's little tight for me. I'll post a make on Printables soon.

Tonight to get the clip on or to tight to get the bin on the grid?

Monniasza, 2025-02-10:
@ostat It's tight both ways. Clip clearance chould be variable both between bins and grid and between grid and clip.