ostat/gridfinity_extended_openscad #79: Vertical Spacing with label shelf to low

MinchinWeb, 2024-12-06:
I've been playing with labels, particularly the legacy Cullenect labels. However, with the label shelf, the bins don't cleanly stack, even without a label installed. The missing clearance is small, probably under 0.5mm. It gets worse when a label is installed, as the default `gflabel` has a 0.4mm embossing (text height). Setting `z clearance` to 1.0 mm solves this, including when labels are installed. Should this z clearance become the default or otherwise added when a label (shelf) is printed?

This sample was printed with 0.2mm layers.

[image]

Zoomed in:

[image]

ostat, 2024-12-06:
Setting an auto min clearance of 1mm sounds like a good idea. Just need to check the impact.

ostat, 2025-02-23:
I know this is an old ticket. Sorry for the delay.

I have changed the default zClearance (renamed to headroom) to 0.8. I found this to be a good setting for my printer.
d5bf791

MinchinWeb, 2025-02-25:
Should do the trick!

closed 2025-02-25 by ostat