ostat/gridfinity_extended_openscad #81: How to generate (sliding) lids?

MinchinWeb, 2024-12-06:
I see how holes for sliding lids can be generated in the bins, but can I generate the actual sliding lids from the collection?

Like what is shown in the picture (from [here](https://github.com/ostat/gridfinity_extended_openscad/issues/47#issuecomment-2464862273)):

![lid](https://private-user-images.githubusercontent.com/2128234/384398268-bc5dc12c-1285-415e-a3f8-6d3514001f4e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM0NTk2ODEsIm5iZiI6MTczMzQ1OTM4MSwicGF0aCI6Ii8yMTI4MjM0LzM4NDM5ODI2OC1iYzVkYzEyYy0xMjg1LTQxNWUtYTNmOC02ZDM1MTQwMDFmNGUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMDZUMDQyOTQxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzhhNTJmODYzMGZlMzQ0ZDQyNmI5ZTQ5ZjJhOTg1MjBiYmU5MGU4MzNmMDllNjU0NmI0OTUxZTBiNmYyODYwNCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.FDciEZ4sU0uQDEqmlYIeLFSPvZ9Jz5yRdm_v6bMViPE)

ostat, 2024-12-06:
There should be an option, something like **sliding lid lip**

ostat, 2024-12-06:
[image]

MinchinWeb, 2024-12-09:
Found it! I need to open the `gridfinity_sliding_lid.scad` file (rather than the "basic cup" one)

closed 2024-12-09 by MinchinWeb