ostat/gridfinity_extended_openscad #86: [BUG] The drawer top plate is the same whether it is default or efficient magnet base

Monniasza, 2024-12-07:
Hi. When I pulled your last commit, the drawers chest always has the efficient magnet base. This happens on both of my repositories.

Monniasza, 2024-12-07:
[image]
[image]

ostat, 2024-12-07:
I rewrote the base script, I will need to update the cheat options to align. 

I replaced the old list of base types with individual options for each feature. 
So you no longer pick the magnet base or weighted base, you now enable or disable the magnets. This allows for more flexibility.

Monniasza, 2024-12-07:
Concerning #83 and #82, I've removed the top plate and base checkboxes and used the list only, with the "None" option disabling top plates and bases respectively.

Monniasza, 2024-12-08:
@ostat After merging the #84, this issue persists.

ostat, 2024-12-08:
That is a different issue than what the resolved. 
It's not fixed yet. 
I will try fix it this week.
I also want to use some of the new features to further improve the way the grid is used in chest. 
I will let you know when it's done so you can try it out.

Monniasza, 2024-12-12:
@ostat Have you fixed it?

ostat, 2024-12-21:
The updated baseplate has been incorporated in to chest,

This fixes the magnet option, fills the space around the baseplate to make it look cleaner and print easier, fixed some issues with the placement of plate in the drawers.

[image]

Monniasza, 2024-12-22:
It's fixed.
[image]

closed 2024-12-23 by ostat