ostat/gridfinity_extended_openscad #88: Default gridfinity_item_holder.scad fails to render

flaviut, 2024-12-16:
Thank you for all your hard work on this, this project looks like it's going to save me a ton of time organizing my coffee supplies.

I ran into issues with the default uncustomized `gridfinity_item_holder.scad`. When I try and render it in openscad, I get the following errors and nothing shows up:

```
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
ECHO: "•••••• UB (USER library v2022) included! v.gd/ubaer  ••••••
• Version: β23.305 v [2024, 12, 15] • Design: 0
Layer: 0.08(24.37%) Nozzle ∅: 0.4 • fn:36 fs:0.75 fa:10 • Spiel: 0.2 •"
ECHO: "❌••••• Help off       use: helpsw=1;  •••••"
ECHO: "Schalter•
 messpunkt=🟢✔ • vp=❌ • anima=❌ • help=❌ • $info=🟢✔ • bed=❌ • hires=❌ •"
ECHO: "
---————————————————————————————————————————---
"
WARNING: Ignoring unknown variable 'magnet_diameter' in file gridfinity_item_holder.scad, line 606
WARNING: Ignoring unknown variable 'screw_depth' in file gridfinity_item_holder.scad, line 608
WARNING: Ignoring unknown variable 'center_magnet_diameter' in file gridfinity_item_holder.scad, line 609
WARNING: Ignoring unknown variable 'center_magnet_thickness' in file gridfinity_item_holder.scad, line 610
WARNING: undefined operation (undefined > number) in file gridfinity_item_holder.scad, line 695
WARNING: max() parameter could not be converted: argument 1: expected number, found undefined (undef) in file modules/functions_gridfinity.scad, line 40
WARNING: max() parameter could not be converted: argument 1: expected number, found undefined (undef) in file modules/functions_gridfinity.scad, line 40
WARNING: undefined operation (undefined + number) in file modules/functions_gridfinity.scad, line 43
WARNING: undefined operation (undefined + number) in file gridfinity_item_holder.scad, line 704
WARNING: undefined operation (undefined / number) in file gridfinity_item_holder.scad, line 704
WARNING: ceil() parameter could not be converted: argument 0: expected number, found undefined (undef) in file gridfinity_item_holder.scad, line 704
WARNING: max() parameter could not be converted: argument 0: expected number, found undefined (undef) in file modules/functions_gridfinity.scad, line 40
ERROR: Assertion 'is_num(magnet_depth)' failed: "magnet_depth must be a number" in file modules/functions_gridfinity.scad, line 49
TRACE: called by 'calculateFloorHeight' in file modules/functions_gridfinity.scad, line 62
TRACE: called by 'calculateFloorThickness' in file gridfinity_item_holder.scad, line 707
TRACE: assignment to 'ft' in file gridfinity_item_holder.scad, line 707
TRACE: called by 'gridfinity_itemholder' in file gridfinity_item_holder.scad, line 827
```

I ran a bisect and 114b7ee is the latest commit that works for me. Tested with the latest openscad git, https://github.com/openscad/openscad/commit/5e8738563c61ec53ff51e995f912f0b57754ecf6

petervk, 2024-12-31:
Same here, thanks for all your great work but also can't get the default item holder to work. Here is the error I get:

```
OpenSCAD 2024.12.30 
https://www.openscad.org/
 
Copyright (C) 2009-2024 The OpenSCAD Developers
This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.
 
Loaded design 'C:/Users/peter/OneDrive/3D Printer/gridfinity_extended_openscad/gridfinity_item_holder.scad'.
Used file cache size: 6 files
Compiling design (CSG Tree generation)...
ECHO: "•••••• UB (USER library v2022) included! v.gd/ubaer  ••••••
• Version: β23.305 v [2024, 12, 30] • Design: 0
Layer: 0.08(24.37%) Nozzle ∅: 0.4 • fn:36 fs:0.75 fa:10 • Spiel: 0.2 •"
ECHO: "❌••••• Help off       use: helpsw=1;  •••••"
ECHO: "Schalter•
 messpunkt=🟢✔ • vp=❌ • anima=❌ • help=❌ • $info=🟢✔ • bed=❌ • hires=❌ •"
ECHO: "
---————————————————————————————————————————---
"
[WARNING: Ignoring unknown variable 'magnet_diameter' in file gridfinity_item_holder.scad, line 606](https://github.com/ostat/gridfinity_extended_openscad/issues/606,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/gridfinity_item_holder.scad)
[WARNING: Ignoring unknown variable 'screw_depth' in file gridfinity_item_holder.scad, line 608](https://github.com/ostat/gridfinity_extended_openscad/issues/608,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/gridfinity_item_holder.scad)
[WARNING: Ignoring unknown variable 'center_magnet_diameter' in file gridfinity_item_holder.scad, line 609](https://github.com/ostat/gridfinity_extended_openscad/issues/609,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/gridfinity_item_holder.scad)
[WARNING: Ignoring unknown variable 'center_magnet_thickness' in file gridfinity_item_holder.scad, line 610](https://github.com/ostat/gridfinity_extended_openscad/issues/610,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/gridfinity_item_holder.scad)
[WARNING: undefined operation (undefined > number) in file gridfinity_item_holder.scad, line 695](https://github.com/ostat/gridfinity_extended_openscad/issues/695,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/gridfinity_item_holder.scad)
[WARNING: max() parameter could not be converted: argument 1: expected number, found undefined (undef) in file modules/functions_gridfinity.scad, line 40](https://github.com/ostat/gridfinity_extended_openscad/issues/40,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/modules/functions_gridfinity.scad)
[WARNING: max() parameter could not be converted: argument 1: expected number, found undefined (undef) in file modules/functions_gridfinity.scad, line 40](https://github.com/ostat/gridfinity_extended_openscad/issues/40,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/modules/functions_gridfinity.scad)
[WARNING: undefined operation (undefined + number) in file modules/functions_gridfinity.scad, line 43](https://github.com/ostat/gridfinity_extended_openscad/issues/43,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/modules/functions_gridfinity.scad)
[WARNING: undefined operation (undefined + number) in file gridfinity_item_holder.scad, line 704](https://github.com/ostat/gridfinity_extended_openscad/issues/704,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/gridfinity_item_holder.scad)
[WARNING: undefined operation (undefined / number) in file gridfinity_item_holder.scad, line 704](https://github.com/ostat/gridfinity_extended_openscad/issues/704,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/gridfinity_item_holder.scad)
[WARNING: ceil() parameter could not be converted: argument 0: expected number, found undefined (undef) in file gridfinity_item_holder.scad, line 704](https://github.com/ostat/gridfinity_extended_openscad/issues/704,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/gridfinity_item_holder.scad)
[WARNING: max() parameter could not be converted: argument 0: expected number, found undefined (undef) in file modules/functions_gridfinity.scad, line 40](https://github.com/ostat/gridfinity_extended_openscad/issues/40,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/modules/functions_gridfinity.scad)
[ERROR: Assertion 'is_num(magnet_depth)' failed: "magnet_depth must be a number" in file modules/functions_gridfinity.scad, line 49](https://github.com/ostat/gridfinity_extended_openscad/issues/49,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/modules/functions_gridfinity.scad)
[TRACE: called by 'calculateFloorHeight' in file modules/functions_gridfinity.scad, line 62](https://github.com/ostat/gridfinity_extended_openscad/issues/62,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/modules/functions_gridfinity.scad)
[TRACE: called by 'calculateFloorThickness' in file gridfinity_item_holder.scad, line 707](https://github.com/ostat/gridfinity_extended_openscad/issues/707,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/gridfinity_item_holder.scad)
[TRACE: assignment to 'ft' in file gridfinity_item_holder.scad, line 707](https://github.com/ostat/gridfinity_extended_openscad/issues/707,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/gridfinity_item_holder.scad)
[TRACE: called by 'gridfinity_itemholder' in file gridfinity_item_holder.scad, line 827](https://github.com/ostat/gridfinity_extended_openscad/issues/827,C:/Users/peter/OneDrive/3D%20Printer/gridfinity_extended_openscad/gridfinity_item_holder.scad)
Compiling design (CSG Products generation)...
Geometries in cache: 0
Geometry cache size in bytes: 0
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized tree has 1 elements!
Compile and preview finished.
Total rendering time: 0:00:00.101
```

ostat, 2025-01-04:
This should now be working

[image]

flaviut, 2025-01-04:
Confirmed to be fixed for me too! thank you so much!

closed 2025-01-04 by ostat

ostat, 2025-01-04:
Thanks for confirming. It's appreciated.

petervk, 2025-01-04:
Yes, this fixed it for me also. Thank you!