ostat/gridfinity_extended_openscad #93: syntax error

tsiawd, 2024-12-25:
I keep getting this error on opening the files.

(C) 2009-2021 The OpenSCAD Developers
This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.

Loaded design 'C:/Users/trevo/Downloads/gridfinity_extended_openscad-main/gridfinity_item_holder.scad'.
[ERROR: Parser error: syntax error in file modules/module_gridfinity_cup_base.scad, line 130](https://github.com/ostat/gridfinity_extended_openscad/issues/130,C:/Users/trevo/Downloads/gridfinity_extended_openscad-main/modules/module_gridfinity_cup_base.scad) 
Execution aborted

Clarkbf, 2024-12-27:
I have the same problem. Except mine is Line 135.
132 let(
133    efficientFloor = validateEfficientFloor(settings[iCupBase_EfficientFloor]),
134    magnetEasyRelease = validateMagnetEasyRelease(settings[iCupBase_MagnetEasyRelease], efficientFloor),
135     ) [

ostat, 2025-01-04:
The item holder script should now be working, there were a lot of changes needed, let me know if I missed anything
[image]
[image]

closed 2025-01-10 by ostat