ostat/gridfinity_extended_openscad #94: Total size versus grid size causes error

CaffeinatedPython, 2024-12-28:
[image]
[image]

ostat, 2024-12-28:
What base plate size are you using?

ostat, 2024-12-28:
I think the issue here is the last plate is too small. This results in issues trying to render it.

One way I have thought to try resolve this is to try an spread out the cell across the plates. Instead of have a plate of 5 then 5 then 0.5 have something like 4 then 4 then 2.5.

I have added a new option to attempt to address this, its not perfect, and it does mean there will likely be more total plates. 

So this
[image]

becomes this
[image]

closed 2025-01-10 by ostat