ostat/gridfinity_extended_openscad #96: Errors in latest when attempting baseplate

dankcushions, 2024-12-29:
```
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
[WARNING: Ignoring unknown function 'calcDimensionWidth' in file gridfinity_baseplate.scad, line 177](https://github.com/ostat/gridfinity_extended_openscad/issues/177,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[WARNING: Ignoring unknown function 'calcDimensionDepth' in file gridfinity_baseplate.scad, line 178](https://github.com/ostat/gridfinity_extended_openscad/issues/178,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[WARNING: Ignoring unknown function 'calcDimensionWidth' in file gridfinity_baseplate.scad, line 179](https://github.com/ostat/gridfinity_extended_openscad/issues/179,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[WARNING: Ignoring unknown function 'calcDimensionWidth' in file gridfinity_baseplate.scad, line 180](https://github.com/ostat/gridfinity_extended_openscad/issues/180,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[ERROR: Assertion 'is_num(gf_size)' failed: "gf_size must be a number" in file gridfinity_baseplate.scad, line 88](https://github.com/ostat/gridfinity_extended_openscad/issues/88,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[TRACE: called by 'split_dimention' in file gridfinity_baseplate.scad, line 124](https://github.com/ostat/gridfinity_extended_openscad/issues/124,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[TRACE: called by 'split_plate' in file gridfinity_baseplate.scad, line 184](https://github.com/ostat/gridfinity_extended_openscad/issues/184,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
[TRACE: called by 'if' in file gridfinity_baseplate.scad, line 164](https://github.com/ostat/gridfinity_extended_openscad/issues/164,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/gridfinity_baseplate.scad) 
Compiling design (CSG Products generation)...
Geometries in cache: 0
Geometry cache size in bytes: 0
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized tree has 1 elements!
Compile and preview finished.
Total rendering time: 0:00:00.137
```

ostat, 2024-12-29:
How did you run this?
Did you download all files?

ostat, 2025-01-01:
I believe this is fixed.

dankcushions, 2025-01-06:
Thanks! I downloaded the latest 'developer snapshot' for openscad (was using the 2021 stable before) and the baseplate now seems to work, but not so much luck with basic cup:

```
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
ECHO: "🟩height: 3gf (21mm)", input = [3, 0], roundedCalcUnits = 3
ECHO: "🟩width: 2gf (84mm)", input = [2, 0], roundedCalcUnits = 2
ECHO: "🟩depth: 1gf (42mm)", input = [1, 0], roundedCalcUnits = 1
ECHO: "🟩height: 3gf (21mm)", input = [3, 0], roundedCalcUnits = 3
[ERROR: Assertion 'is_bool(settings[iCupBase_FlatBase])' failed: "CupBase FlatBase Settings must be a boolean" in file modules/module_gridfinity_cup_base.scad, line 128](https://github.com/ostat/gridfinity_extended_openscad/issues/128,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/modules/module_gridfinity_cup_base.scad)
[TRACE: called by 'ValidateCupBaseSettings' in file modules/module_gridfinity_cup_base.scad, line 115](https://github.com/ostat/gridfinity_extended_openscad/issues/115,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/modules/module_gridfinity_cup_base.scad)
[TRACE: called by 'CupBaseSettings' in file gridfinity_basic_cup.scad, line 252](https://github.com/ostat/gridfinity_extended_openscad/issues/252,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/gridfinity_basic_cup.scad)
[TRACE: called by 'gridfinity_cup' in file gridfinity_basic_cup.scad, line 239](https://github.com/ostat/gridfinity_extended_openscad/issues/239,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/gridfinity_basic_cup.scad)
[TRACE: called by 'children' in file modules/functions_general.scad, line 152](https://github.com/ostat/gridfinity_extended_openscad/issues/152,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/modules/functions_general.scad)
[TRACE: called by 'difference' in file modules/functions_general.scad, line 150](https://github.com/ostat/gridfinity_extended_openscad/issues/150,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/modules/functions_general.scad)
[TRACE: called by 'union' in file modules/functions_general.scad, line 149](https://github.com/ostat/gridfinity_extended_openscad/issues/149,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/modules/functions_general.scad)
[TRACE: called by 'translate' in file modules/functions_general.scad, line 148](https://github.com/ostat/gridfinity_extended_openscad/issues/148,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/modules/functions_general.scad)
[TRACE: call of 'SetGridfinityEnvironment(width = [2, 0], depth = [1, 0], height = [3, 0], setColour = "enable", help = "disabled", render_position = "center", cutx = 0, cuty = 0, cutz = 3)' in file modules/functions_general.scad, line 128](https://github.com/ostat/gridfinity_extended_openscad/issues/128,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/modules/functions_general.scad)
[TRACE: called by 'SetGridfinityEnvironment' in file gridfinity_basic_cup.scad, line 229](https://github.com/ostat/gridfinity_extended_openscad/issues/229,/Users/chris/Documents/OpenSCAD/libraries/gridfinity_extended_openscad/gridfinity_basic_cup.scad)
Compiling design (CSG Products generation)...
Geometries in cache: 0
Geometry cache size in bytes: 0
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized tree has 1 elements!
Compile and preview finished.
Total rendering time: 0:00:00.170
```

Doesn't seem to matter if I go to <design default> for the params (or any other). fiddling with the options for base > flat base doesn't seem to make a difference.

Very possible I am installing/running this wrong! First dabble with OpenSCAD. just cloned main to 
`Documents/OpenSCAD/libraries/gridfinity_extended_openscad/` and then opened `gridfinity_basic_cup.scad`. This is on OSX.

ostat, 2025-01-11:
Thanks for raising.
This was a different issue and hows now been fixed. There are details in this ticket  #105.