ostat/gridfinity_extended_openscad #98: Feature Request - Extension Tabs on Dividers & Finger Slide Height Adjustment

Raatner, 2024-12-29:
Hello! Just want to start by saying - after finding and testing probably a dozen parametric Gridfinity 'generators' to overhaul our kitchen drawers, yours have been the most approachable and enjoyable to work with by far, thank you for all your work on this <3

One thing I've been consistently changing in the slicer before printing has been to use the cut tool to copy the extension tabs from the outer walls to the inner dividers. I'm not sure what would be involved in having them added in the generated model, but thought it might be a neat feature as I did find it helped a lot with rigidity and keeping things aligned.

I also noticed that finger slides beyond radius 5 or 6 appear on the model that they will interfere with extension tabs. Thought it might be possible to calculate a Z offset based on slide radius and wall height to automatically adjust them up above the top of the slide, or maybe an extra field to manually set it since my brain hurts just thinking about that math.

Again, thank you for everything you've already done - this whole project became completely painless after finding yours, and the wife has been loving how quickly things have progressed because of it :)

[image]

ostat, 2024-12-29:
Having the tabs on the divider walls is a good idea.   It will be hard to do it for bent walls but if they are not bent should be ok.

ostat, 2025-02-15:
@Raatner , I have not fixed this yet. However I added another feature I thought you might like to know about. The finger slide radius is now set per side. So you can have large radius on the sides, and smaller on the end if you want.