ostat/gridfinity_extended_openscad #100: gridfinity_basic_cup.scad with cullenect V2 labels seems to chop one side of the lip completely off

ids444, 2024-12-30:
[image]

gridfinity_basic_cup.scad with cullenect V2 labels seems to chop one side of the lip completely off.

Doesn't seem to matter what other setting I change, lip styles, box dimension, etc. Label position changes where the cutout happens, with centered it will cutout both sides.

This doesn't happen with the online version at makerworld, can't test now as they are down for maintenance.

timpiglowski, 2025-01-23:
Is there any generator with support for cullenect V2 right now?

ostat, 2025-01-24:
This has been fixed.
Feedback welcome

[image]

ids444, 2025-01-24:
Thanks! Confirmed on my end

closed 2025-01-24 by ids444

ids444, 2025-01-24:
> Is there any generator with support for cullenect V2 right now?

Yes, there is gflabel

https://github.com/ndevenish/gflabel