ostat/gridfinity_extended_openscad #101: Several issues after item holder refactoring

shrestha-pranav, 2024-12-31:
Seems like `gridfinity_item_holder` is completely broken right now. This is on the latest commit on main using the default build parameters, running `OpenSCAD Version: 2024.09.20 (git 4b645fc30)`

```
OpenSCAD 2024.09.20 
https://www.openscad.org/
 
Copyright (C) 2009-2024 The OpenSCAD Developers
This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.
 
Loaded design '<path>/gridfinity_extended_openscad/gridfinity_item_holder.scad'.
Used file cache size: 6 files
Compiling design (CSG Tree generation)...
ECHO: "•••••• UB (USER library v2022) included! v.gd/ubaer  ••••••
• Version: β23.305 v [2024, 9, 20] • Design: 0
Layer: 0.08(24.37%) Nozzle ∅: 0.4 • fn:36 fs:0.75 fa:10 • Spiel: 0.2 •"
ECHO: "❌••••• Help off       use: helpsw=1;  •••••"
ECHO: "Schalter•
 messpunkt=🟢✔ • vp=❌ • anima=❌ • help=❌ • $info=🟢✔ • bed=❌ • hires=❌ •"
ECHO: "
---————————————————————————————————————————---
"
WARNING: Ignoring unknown variable 'magnet_diameter' in file gridfinity_item_holder.scad, line 606
WARNING: Ignoring unknown variable 'screw_depth' in file gridfinity_item_holder.scad, line 608
WARNING: Ignoring unknown variable 'center_magnet_diameter' in file gridfinity_item_holder.scad, line 609
WARNING: Ignoring unknown variable 'center_magnet_thickness' in file gridfinity_item_holder.scad, line 610
WARNING: undefined operation (undefined > number) in file gridfinity_item_holder.scad, line 695
WARNING: max() parameter could not be converted: argument 1: expected number, found undefined (undef) in file modules/functions_gridfinity.scad, line 40
WARNING: max() parameter could not be converted: argument 1: expected number, found undefined (undef) in file modules/functions_gridfinity.scad, line 40
WARNING: undefined operation (undefined + number) in file modules/functions_gridfinity.scad, line 43
WARNING: undefined operation (undefined + number) in file gridfinity_item_holder.scad, line 704
WARNING: undefined operation (undefined / number) in file gridfinity_item_holder.scad, line 704
WARNING: ceil() parameter could not be converted: argument 0: expected number, found undefined (undef) in file gridfinity_item_holder.scad, line 704
ERROR: Assertion 'is_num(magnet_depth)' failed: "magnet_depth must be a number" in file modules/functions_gridfinity.scad, line 49
TRACE: called by 'calculateFloorHeight' in file modules/functions_gridfinity.scad, line 62
TRACE: called by 'calculateFloorThickness' in file gridfinity_item_holder.scad, line 707
TRACE: assignment to 'ft' in file gridfinity_item_holder.scad, line 707
TRACE: called by 'gridfinity_itemholder' in file gridfinity_item_holder.scad, line 827
Compiling design (CSG Products generation)...
Geometries in cache: 0
Geometry cache size in bytes: 0
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized tree has 1 elements!
Compile and preview finished.
Total rendering time: 0:00:00.041
 
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
ECHO: "•••••• UB (USER library v2022) included! v.gd/ubaer  ••••••
• Version: β23.305 v [2024, 9, 20] • Design: 0
Layer: 0.08(24.37%) Nozzle ∅: 0.4 • fn:72 fs:0.2 fa:1 • Spiel: 0.2 •"
ECHO: "❌••••• Help off       use: helpsw=1;  •••••"
ECHO: "Schalter•
 messpunkt=❌ • vp=🟢✔ • anima=❌ • help=❌ • $info=❌ • bed=❌ • hires=❌ •"
ECHO: "
    Viewportcontrol vpr: [55, 0, 25]
    Viewportcontrol vpt: [0, 0, 0]
    Viewportcontrol vpd: 140
    Viewportcontrol vpf: 22.5
    ••••  vp=on  •••••"
ECHO: "
        ⏳ Rendering…wait! ⌛"
ECHO: "
---————————————————————————————————————————---
"
WARNING: Ignoring unknown variable 'magnet_diameter' in file gridfinity_item_holder.scad, line 606
WARNING: Ignoring unknown variable 'screw_depth' in file gridfinity_item_holder.scad, line 608
WARNING: Ignoring unknown variable 'center_magnet_diameter' in file gridfinity_item_holder.scad, line 609
WARNING: Ignoring unknown variable 'center_magnet_thickness' in file gridfinity_item_holder.scad, line 610
WARNING: undefined operation (undefined > number) in file gridfinity_item_holder.scad, line 695
WARNING: max() parameter could not be converted: argument 1: expected number, found undefined (undef) in file modules/functions_gridfinity.scad, line 40
WARNING: max() parameter could not be converted: argument 1: expected number, found undefined (undef) in file modules/functions_gridfinity.scad, line 40
WARNING: undefined operation (undefined + number) in file modules/functions_gridfinity.scad, line 43
WARNING: undefined operation (undefined + number) in file gridfinity_item_holder.scad, line 704
WARNING: undefined operation (undefined / number) in file gridfinity_item_holder.scad, line 704
WARNING: ceil() parameter could not be converted: argument 0: expected number, found undefined (undef) in file gridfinity_item_holder.scad, line 704
ERROR: Assertion 'is_num(magnet_depth)' failed: "magnet_depth must be a number" in file modules/functions_gridfinity.scad, line 49
TRACE: called by 'calculateFloorHeight' in file modules/functions_gridfinity.scad, line 62
TRACE: called by 'calculateFloorThickness' in file gridfinity_item_holder.scad, line 707
TRACE: assignment to 'ft' in file gridfinity_item_holder.scad, line 707
TRACE: called by 'gridfinity_itemholder' in file gridfinity_item_holder.scad, line 827
Rendering Polygon Mesh using Manifold...
WARNING: No top level geometry to render
```

From a rough look, a lot of variables are undefined after a previous refactoring. Not sure if this is a known issue.

Are there any known good commits I can revert to?

shrestha-pranav, 2024-12-31:
Seems like `combined/gridfinity_item_holder.scad` is the new `gridfinity_item_holder.scad`. Is this a temporary change? If not, maybe removing the original file or documenting this somewhere may be better :)

shrestha-pranav, 2025-01-02:
PR #69 and #70 probably fix the issues here. Now I see this error though. Should I stick to `combined/`? Any insights would be greatly appreciated.

```
OpenSCAD 2024.09.20 
https://www.openscad.org/
 
Copyright (C) 2009-2024 The OpenSCAD Developers
This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.
 
Loaded design '/Users/ps2958/3d/gridfinity_extended_openscad/gridfinity_item_holder.scad'.
Used file cache size: 6 files
Compiling design (CSG Tree generation)...
ECHO: "•••••• UB (USER library v2022) included! v.gd/ubaer  ••••••
• Version: β23.305 v [2024, 9, 20] • Design: 0
Layer: 0.08(24.37%) Nozzle ∅: 0.4 • fn:36 fs:0.75 fa:10 • Spiel: 0.2 •"
ECHO: "❌••••• Help off       use: helpsw=1;  •••••"
ECHO: "Schalter•
 messpunkt=🟢✔ • vp=❌ • anima=❌ • help=❌ • $info=🟢✔ • bed=❌ • hires=❌ •"
ECHO: "
---————————————————————————————————————————---
"
[WARNING: variable position not specified as parameter in file gridfinity_item_holder.scad, line 734](https://github.com/ostat/gridfinity_extended_openscad/issues/734,/Users/ps2958/3d/gridfinity_extended_openscad/gridfinity_item_holder.scad)
ECHO: "🟩width: 2gf (84mm)", input = [2, 0], roundedCalcUnits = 2
ECHO: "🟩depth: 1gf (42mm)", input = [1, 0], roundedCalcUnits = 1
ECHO: "🟩height: 2.1gf (14.7mm)", input = 2.1
ECHO: "cutout_pattern", patternVariable = 0.75
ECHO: " true Grid Gridsize([2, 2, 1])=4 elements= 13.547×12×0mm 
 element spacing= [13.547, 12, 0] mm
	X -6.7735 ⇔ 6.7735 mm
	Y -6 ⇔ 6 mm
	Z 0 ⇔ 0 mm"
ECHO: "cutout_pattern", patternVariable = 0.75
ECHO: " true Grid Gridsize([2, 2, 1])=4 elements= 13.547×12×0mm 
 element spacing= [13.547, 12, 0] mm
	X -6.7735 ⇔ 6.7735 mm
	Y -6 ⇔ 6 mm
	Z 0 ⇔ 0 mm"
[WARNING: Ignoring unknown variable '$num_x' in file modules/module_gridfinity_cup_base_text.scad, line 51](https://github.com/ostat/gridfinity_extended_openscad/issues/51,/Users/ps2958/3d/gridfinity_extended_openscad/modules/module_gridfinity_cup_base_text.scad)
[WARNING: Ignoring unknown variable '$num_y' in file modules/module_gridfinity_cup_base_text.scad, line 53](https://github.com/ostat/gridfinity_extended_openscad/issues/53,/Users/ps2958/3d/gridfinity_extended_openscad/modules/module_gridfinity_cup_base_text.scad)
[WARNING: Ignoring unknown variable '$num_z' in file modules/module_gridfinity_cup_base_text.scad, line 55](https://github.com/ostat/gridfinity_extended_openscad/issues/55,/Users/ps2958/3d/gridfinity_extended_openscad/modules/module_gridfinity_cup_base_text.scad)
[WARNING: Experimental builtin function 'textmetrics' is not enabled in file modules/module_gridfinity_cup_base_text.scad, line 59](https://github.com/ostat/gridfinity_extended_openscad/issues/59,/Users/ps2958/3d/gridfinity_extended_openscad/modules/module_gridfinity_cup_base_text.scad)
[WARNING: Ignoring unknown function 'textmetrics' in file modules/module_gridfinity_cup_base_text.scad, line 59](https://github.com/ostat/gridfinity_extended_openscad/issues/59,/Users/ps2958/3d/gridfinity_extended_openscad/modules/module_gridfinity_cup_base_text.scad)
[WARNING: undefined operation (number / undefined) in file modules/module_gridfinity_cup_base_text.scad, line 60](https://github.com/ostat/gridfinity_extended_openscad/issues/60,/Users/ps2958/3d/gridfinity_extended_openscad/modules/module_gridfinity_cup_base_text.scad)
ECHO: " true Grid Gridsize([10, 4, 1])=40 elements= 67.0304×25.8×0mm 
 element spacing= [7.44782, 8.6, 0] mm
	X -33.5152 ⇔ 33.5152 mm
	Y -12.9 ⇔ 12.9 mm
	Z 0 ⇔ 0 mm"
ECHO: " true HexGrid elements=35"
Compiling design (CSG Products generation)...
Geometries in cache: 91
Geometry cache size in bytes: 711840
CGAL Polyhedrons in cache: 30
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized tree has 478 elements!
Compile and preview finished.
Total rendering time: 0:00:00.240
```

Maybe I need to update my OpenSCAD version.

ostat, 2025-01-02:
Hi, yes sorry I broke it. 
I should have a fix up later today, I have it mostly resolved but there was an alignment issue I need to resolve. 

If you need it sooner, try the combined script, or the version on makerworld. As this might be a bit older. 

Sorry for the inconvenience.

ostat, 2025-01-04:
The item holder script should now be working, there were a lot of changes needed, let me know if I missed anything
[image]
[image]

closed 2025-01-10 by ostat

akavlie, 2025-01-11:
I'm having issues with this script; it often breaks depending on the item I'm trying to render or the size.
For instance, if I increase the depth past 1.0, it fails to render.

I'm using the combined version, generated 2025-01-04 at 10:13.