ostat/gridfinity_extended_openscad #105: flatBase is bool or string?

danielsan80, 2025-01-09:
https://github.com/ostat/gridfinity_extended_openscad/blob/ca2a82c4c8515291d146d8c14b80aa34eb4c001c/gridfinity_basic_cup.scad#L274

`flat_base = "off"; // [off, gridfinity, rounded]`

and

` flatBase=flat_base,`

but

` assert(is_bool(settings[iCupBase_FlatBase]), "CupBase FlatBase Settings must be a boolean")`

so maybe

`flatBase=flat_base!="off",`

ostat, 2025-01-09:
I did mean to check that in. It's for a new change I am working on. 

For. Ow you can change it back to a Boolean. 

'flat_base = false;'

ostat, 2025-01-10:
This should be fixed now, 
The flat base option has changed from a boolean to a dropdown, of `off`, `gridfinity` and `rounded`.
`gridfinity` add a gridfinity base around the edges, rounded add a rounded bottom edge.

closed 2025-01-24 by ostat