ostat/gridfinity_extended_openscad #107: Error when applying wall pattern to dividers

akavlie, 2025-01-11:
I get an error when I try to apply a wall pattern to a divider. The bin fails to render.

Version: `Combined version of 'gridfinity_basic_cup.scad'. Generated 2025-01-04 09:07`

Console logs:

```
Parsing design (AST generation)...
Compiling design (CSG Tree generation)...
ECHO: "•••••• UB (USER library v2022) included! v.gd/ubaer  ••••••
• Version: β23.305 v [2025, 1, 10] • Design: 0
Layer: 0.08(24.37%) Nozzle ∅: 0.4 • fn:36 fs:0.75 fa:10 • Spiel: 0.2 •"
ECHO: "❌••••• Help off       use: helpsw=1;  •••••"
ECHO: "Schalter•
 messpunkt=🟢✔ • vp=❌ • anima=❌ • help=❌ • $info=🟢✔ • bed=❌ • hires=❌ •"
ECHO: "
---————————————————————————————————————————---
"
ECHO: "🟩height: 12gf (84mm)", input = [12, 0], roundedCalcUnits = 12
ECHO: "🟩width: 1gf (42mm)", input = [1, 0], roundedCalcUnits = 1
ECHO: "🟩depth: 1gf (42mm)", input = [1, 0], roundedCalcUnits = 1
ECHO: "🟩height: 12gf (84mm)", input = [12, 0], roundedCalcUnits = 12
ECHO: "🟩width: 1gf (42mm)", input = [1, 0], roundedCalcUnits = 1
ECHO: "🟩depth: 1gf (42mm)", input = [1, 0], roundedCalcUnits = 1
ECHO: "🟩height: 12gf (84mm)", input = [12, 0], roundedCalcUnits = 12
ECHO: "cutout_pattern", patternVariable = 0.75
ECHO: " true Grid Gridsize([14, 6, 1])=84 elements= 78.8083×35×0mm 
 element spacing= [6.06218, 7, 0] mm
	X -39.4042 ⇔ 39.4042 mm
	Y -17.5 ⇔ 17.5 mm
	Z 0 ⇔ 0 mm"
ECHO: "cutout_pattern", patternVariable = 0.75
ECHO: " true Grid Gridsize([14, 6, 1])=84 elements= 78.8083×35×0mm 
 element spacing= [6.06218, 7, 0] mm
	X -39.4042 ⇔ 39.4042 mm
	Y -17.5 ⇔ 17.5 mm
	Z 0 ⇔ 0 mm"
ECHO: "cutout_pattern", patternVariable = 0.75
ECHO: " true Grid Gridsize([14, 6, 1])=84 elements= 78.8083×35×0mm 
 element spacing= [6.06218, 7, 0] mm
	X -39.4042 ⇔ 39.4042 mm
	Y -17.5 ⇔ 17.5 mm
	Z 0 ⇔ 0 mm"
ECHO: "cutout_pattern", patternVariable = 0.75
ECHO: " true Grid Gridsize([14, 6, 1])=84 elements= 78.8083×35×0mm 
 element spacing= [6.06218, 7, 0] mm
	X -39.4042 ⇔ 39.4042 mm
	Y -17.5 ⇔ 17.5 mm
	Z 0 ⇔ 0 mm"
[WARNING: Ignoring unknown variable "left" in file gridfinitybasiccup.scad, line 947](https://github.com/ostat/gridfinity_extended_openscad/issues/947,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: Unable to convert translate(undef) parameter to a vec3 or vec2 of numbers in file gridfinitybasiccup.scad, line 947](https://github.com/ostat/gridfinity_extended_openscad/issues/947,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: Ignoring unknown variable "left" in file gridfinitybasiccup.scad, line 948](https://github.com/ostat/gridfinity_extended_openscad/issues/948,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: Problem converting rotate(a=undef) parameter in file gridfinitybasiccup.scad, line 948](https://github.com/ostat/gridfinity_extended_openscad/issues/948,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: Ignoring unknown variable "left" in file gridfinitybasiccup.scad, line 954](https://github.com/ostat/gridfinity_extended_openscad/issues/954,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: variable "voronoiNoise" not specified as parameter in file gridfinitybasiccup.scad, line 952](https://github.com/ostat/gridfinity_extended_openscad/issues/952,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: variable "voronoiRadius" not specified as parameter in file gridfinitybasiccup.scad, line 952](https://github.com/ostat/gridfinity_extended_openscad/issues/952,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
ECHO: "cutout_pattern", patternVariable = 0
[WARNING: undefined operation (undefined - number) in file gridfinitybasiccup.scad, line 2111](https://github.com/ostat/gridfinity_extended_openscad/issues/2111,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: undefined operation (undefined / number) in file gridfinitybasiccup.scad, line 2111](https://github.com/ostat/gridfinity_extended_openscad/issues/2111,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: undefined operation (undefined + number) in file gridfinitybasiccup.scad, line 2111](https://github.com/ostat/gridfinity_extended_openscad/issues/2111,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: floor() parameter could not be converted: argument 0: expected number, found undefined (undef) in file gridfinitybasiccup.scad, line 2111](https://github.com/ostat/gridfinity_extended_openscad/issues/2111,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: undefined operation (undefined + number) in file gridfinitybasiccup.scad, line 2113](https://github.com/ostat/gridfinity_extended_openscad/issues/2113,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: undefined operation (undefined / number) in file gridfinitybasiccup.scad, line 2113](https://github.com/ostat/gridfinity_extended_openscad/issues/2113,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: undefined operation (undefined - number) in file gridfinitybasiccup.scad, line 2113](https://github.com/ostat/gridfinity_extended_openscad/issues/2113,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: undefined operation (undefined * number) in file gridfinitybasiccup.scad, line 2113](https://github.com/ostat/gridfinity_extended_openscad/issues/2113,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: floor() parameter could not be converted: argument 0: expected number, found undefined (undef) in file gridfinitybasiccup.scad, line 2113](https://github.com/ostat/gridfinity_extended_openscad/issues/2113,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: undefined operation (undefined / number) in file gridfinitybasiccup.scad, line 2113](https://github.com/ostat/gridfinity_extended_openscad/issues/2113,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: undefined operation (undefined + number) in file gridfinitybasiccup.scad, line 2119](https://github.com/ostat/gridfinity_extended_openscad/issues/2119,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: undefined operation (undefined / number) in file gridfinitybasiccup.scad, line 2119](https://github.com/ostat/gridfinity_extended_openscad/issues/2119,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: floor() parameter could not be converted: argument 0: expected number, found undefined (undef) in file gridfinitybasiccup.scad, line 2119](https://github.com/ostat/gridfinity_extended_openscad/issues/2119,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[... 7 lines cut ...]
[WARNING: ceil() parameter could not be converted: argument 0: expected number, found undefined (undef) in file gridfinitybasiccup.scad, line 2129](https://github.com/ostat/gridfinity_extended_openscad/issues/2129,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[WARNING: undefined operation (undefined * undefined) in file gridfinitybasiccup.scad, line 2130](https://github.com/ostat/gridfinity_extended_openscad/issues/2130,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[ERROR: Assertion '(is_list(canvasSize) && (len(canvasSize) == 2))' failed: "canvasSize must be list of len 2" in file gridfinitybasiccup.scad, line 2059](https://github.com/ostat/gridfinity_extended_openscad/issues/2059,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[TRACE: called by 'assert' in file gridfinitybasiccup.scad, line 2059](https://github.com/ostat/gridfinity_extended_openscad/issues/2059,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
[TRACE: call of 'GridItemHolder(canvasSize = undef, hexGrid = true, customShape = false, circleFn = 6, holeSize = [5, 5], holeSpacing = [2, 2], holeGrid = [0, 0], holeHeight = 1.22, holeChamfer = 0, border = 0, center = true, fill = "crop", rotateGrid = true, help = false)' in file gridfinitybasiccup.scad, line 2057](https://github.com/ostat/gridfinity_extended_openscad/issues/2057,/Users [... cut ...]
[TRACE: called by 'translate' in file gridfinitybasiccup.scad, line 946](https://github.com/ostat/gridfinity_extended_openscad/issues/946,/Users/akavlie/Downloads/Gridfinity+Extended+-+Official/gridfinitybasiccup.scad)
Compiling design (CSG Products generation)...
Geometries in cache: 176177
Geometry cache size in bytes: 52937840
CGAL Polyhedrons in cache: 289
CGAL cache size in bytes: 100293168
Compiling design (CSG Products normalization)...
Normalized tree has 1 elements!
Compile and preview finished.
Total rendering time: 0:00:00.445
```

ostat, 2025-01-24:
This error appears to be fixed, however there is a layout error for the dividers.

[image]

ostat, 2025-02-15:
This is now resolved.

closed 2025-05-04 by ostat