ostat/gridfinity_extended_openscad #108: Labels on all sub-bins?

jpypi, 2025-01-11:
I know it's possible to get label tabs on all the sub-bins if there are only columns, but is there a way to add label tabs to all the sub bins when there are multiple rows (Y axis, I think)? Not sure if I'm just missing it.

ostat, 2025-01-13:
Its currently not possible, but is a good idea. 
Gridfinity rebuilt can do that.

ostat, 2025-01-14:
I did consider adding this once before. Due to the way the cavity is created, its not easy to do, but is do able. The bent walls would complicate things, so that would likely be out of scope.

After some consideration, I dropped the idea as I thought that creating two separate bins gives close to the same functionality. I do realise its not exactly the same.

Perhaps a simpler solution would be to support multiple cavities, instead of a single cavity with dividers.

nkiest, 2026-01-17:
Not quite working for me, not sure if a different setting is wrong, but the divider labels don't quite line up with the dividers themselves. Also, they end up with rather different sizes than the ones at the edges.
[image]