ostat/gridfinity_extended_openscad #109: Add upper and lower brim widths

silversword411, 2025-01-16:
I'm trying to add some strength to larger bin prints. Is there a setting to be able to add brim widths on top/bottom/left/right of walls?

[image]

You can see example here trying to make a bin similar but gridfinity compatible https://makerworld.com/en/models/606320?from=search#profileId-529149

ostat, 2025-01-16:
Hi, the wall pattern border/brim is based on the wall thickness. There is currently no setting to override that. 

It something that can be added in a future version.

silversword411, 2025-01-17:
> It something that can be added in a future version.

I look forward to it. 

Chatgpt started me down the right path...but all the `include`s loading chunks from different files I wasn't sure where to add the last bit.

https://chatgpt.com/share/6789f316-4e6c-8013-a444-42ca20f9e720

ostat, 2025-01-17:
The ChatGPT answer is on the right path. 
We need to creat a new array variable, with three values. Top bottom and sides. 
This should be added to the wall pattern object. 
Use that to reducing and positioning the wallpattern. We also need to use it where other objects intersect with the wall pattern like cutouts and dividers.

ostat, 2025-02-15:
This has been added. It allows a border to be added to the wall patterns. It does not yet increase the border where there are cutouts, this is still based on wall_thickness and will require more work.

[image]

ostat, 2025-02-17:
Fixed the cutouts and rounded corners 

[image]

silversword411, 2025-02-17:
Sweet, will do some testing soon thx!

I had actually tried myself, 2 failed attempts later....figured I'd leave it in smarter hands than mine :)

ostat, 2025-02-17:
It's a lot easier when you know the code. 
Let me know what you think.

silversword411, 2025-02-18:
Beeeeeeeautiful!

[image]

closed 2025-02-18 by silversword411

silversword411, 2025-02-18:
FYI still tweaking, but here's the preset I've got to, going to print a couple out and monitor for strength etc.

I like the top having the little bumps that align a top stacker

```
        "Bin 4x6x16 v5": {
            "box_corner_attachments_only": "true",
            "cavity_floor_radius": "-1",
            "center_magnet_size": "[0, 0]",
            "chamber_wall_thickness": "1.2",
            "chamber_wall_zClearance": "0",
            "cutx": "0",
            "cuty": "0",
            "depth": "[6, 0]",
            "efficient_floor": "off",
            "enable_help": "disabled",
            "enable_magnets": "true",
            "enable_screws": "true",
            "extension_tab_size": "[10, 0, 0, 0]",
            "extension_tabs_enabled": "true",
            "extension_x_enabled": "disabled",
            "extension_x_position": "0.5",
            "extension_y_enabled": "disabled",
            "extension_y_position": "0.5",
            "fa": "6",
            "filled_in": "disabled",
            "fingerslide": "none",
            "fingerslide_lip_aligned": "true",
            "fingerslide_radius": "8",
            "fingerslide_walls": "[1, 0, 0, 0]",
            "flat_base": "off",
            "flat_base_rounded_easyPrint": "-1",
            "flat_base_rounded_radius": "-1",
            "floor_thickness": "0.69999999999999996",
            "floorpattern_border": "0",
            "floorpattern_enabled": "false",
            "floorpattern_fill": "crop",
            "floorpattern_hole_radius": "0.5",
            "floorpattern_hole_sides": "6",
            "floorpattern_hole_size": "[5, 5]",
            "floorpattern_hole_spacing": "2",
            "floorpattern_pattern_brick_weight": "5",
            "floorpattern_pattern_grid_chamfer": "0",
            "floorpattern_pattern_quality": "0.40000000000000002",
            "floorpattern_pattern_voronoi_noise": "0.75",
            "floorpattern_style": "gridrotated",
            "fn": "0",
            "force_render": "true",
            "fs": "0.40000000000000002",
            "half_pitch": "false",
            "height": "[12, 0]",
            "hole_overhang_remedy": "2",
            "horizontal_chambers": "1",
            "horizontal_irregular_subdivisions": "false",
            "horizontal_separator_bend_angle": "0",
            "horizontal_separator_bend_position": "0",
            "horizontal_separator_bend_separation": "0",
            "horizontal_separator_config": "10.5|21|42|50|60",
            "horizontal_separator_cut_depth": "0",
            "label_position": "left",
            "label_relief": "[0, 0, 0, 0.6]",
            "label_size": "[0, 14, 0, 0.6]",
            "label_style": "cullenect_legacy",
            "label_walls": "[0, 1, 0, 0]",
            "lip_side_relief_trigger": "[1, 1]",
            "lip_style": "normal",
            "lip_top_notches": "true",
            "lip_top_relief_height": "0",
            "magnet_easy_release": "auto",
            "magnet_size": "[6.5, 2.4]",
            "minimum_printable_pad_size": "0.20000000000000001",
            "pitch": "[42, 42, 7]",
            "random_seed": "0",
            "render_position": "center",
            "screw_size": "[3, 6]",
            "set_colour": "enable",
            "sliding_clearance": "0.10000000000000001",
            "sliding_lid_enabled": "false",
            "sliding_lid_lip_enabled": "false",
            "sliding_lid_thickness": "0",
            "sliding_min_support": "0",
            "sliding_min_wallThickness": "0",
            "spacer": "false",
            "tapered_corner": "none",
            "tapered_corner_size": "10",
            "tapered_setback": "-1",
            "text_1": "false",
            "text_2": "false",
            "text_2_text": "Gridfinity Extended",
            "text_depth": "0.29999999999999999",
            "text_font": "Aldo",
            "text_size": "0",
            "vertical_chambers": "1",
            "vertical_irregular_subdivisions": "false",
            "vertical_separator_bend_angle": "0",
            "vertical_separator_bend_position": "0",
            "vertical_separator_bend_separation": "0",
            "vertical_separator_config": "10.5|21|42|50|60",
            "vertical_separator_cut_depth": "0",
            "wall_thickness": "2.2000000000000002",
            "wallcutout_horizontal": "disabled",
            "wallcutout_horizontal_angle": "70",
            "wallcutout_horizontal_corner_radius": "10",
            "wallcutout_horizontal_height": "50",
            "wallcutout_horizontal_position": "-2",
            "wallcutout_horizontal_width": "0",
            "wallcutout_vertical": "enabled",
            "wallcutout_vertical_angle": "70",
            "wallcutout_vertical_corner_radius": "5",
            "wallcutout_vertical_height": "20",
            "wallcutout_vertical_position": "-2",
            "wallcutout_vertical_width": "55",
            "wallpattern_border": "12",
            "wallpattern_dividers_enabled": "disabled",
            "wallpattern_enabled": "true",
            "wallpattern_fill": "space",
            "wallpattern_hole_radius": "0.5",
            "wallpattern_hole_sides": "6",
            "wallpattern_hole_size": "[6, 5]",
            "wallpattern_hole_spacing": "2",
            "wallpattern_pattern_brick_weight": "7",
            "wallpattern_pattern_grid_chamfer": "0",
            "wallpattern_pattern_quality": "0.40000000000000002",
            "wallpattern_pattern_voronoi_noise": "0.75",
            "wallpattern_style": "hexgridrotated",
            "wallpattern_walls": "[1, 1, 1, 1]",
            "width": "[4, 0]",
            "zClearance": "0"
        }
```