ostat/gridfinity_extended_openscad #112: Superlight baseplate

bobosch, 2025-01-20:
An option to remove the sharp edge on the bottom of the baseplate would be cool
(left: example | right: base plate from this project)
[image]

ostat, 2025-01-20:
Assuming you don't need the other baseplate option. You might be happy with the cnc plate. It does not have the angle sides.

bobosch, 2025-01-20:
I like the superlight baseplates [Baseplate - Superlight.stl.zip](https://github.com/user-attachments/files/18482752/Baseplate.-.Superlight.stl.zip) 
(Fast to print, Less filament, but boxes can move a little bit)

And I like the idea connecting to plates with butterflies...

But I recognizes after printing both that the walls are smaller (that is the reason why the boxes can move a little bit) and so      the butterflies wont be compatible - so it make no sense...

closed 2025-01-20 by bobosch