ostat/gridfinity_extended_openscad #117: [Feature Request] Bins with Half-Sized Base

apetitjean, 2025-01-30:
Hi,

Maybe I just haven't found the functionality, but it seems there's no way to generate bins with a half-sized base.
I printed a grid with half-size cells on the border, but I can't design bins that fit in.

Is there a way to do this, or is it a current limitation of the project?

ostat, 2025-01-30:
Under the base settings there will a an option called half pitch.

https://docs.ostat.com/docs/openscad/gridfinity-extended/basic-cup/#base

apetitjean, 2025-01-30:
Thank you for your answer but this doesn't work for my case. 

I opened a thread on Reddit and I joined a picture of my grid. You will understand as soon as you see it : https://www.reddit.com/r/gridfinity/comments/1idj13f/comment/m9zed1t/?context=3

ostat, 2025-01-30:
Oh I see, you are after the base plate in half pitch, I was thinking the base of the cup. 

This feature is not available at the moment. It should not be needed. Full pitch and Half pitch bins fit on the standard base plate. 
A halftime pitch plate will break compatibility, where a ladle pitch bin does not break compatibility and allows offsetting the bins. 

That being said I would like to expose the pitch sizes (42mm and 7mm) to be configurable. This will be a while as the constants are used in a lot of places.

ostat, 2025-01-30:
I'm not sure I follow. Gridfinity extended supports half pitch bins. 
You can use the official online generator at maker world which has all the options. 

Perplexinglabs only supports a subset of features.

ostat, 2025-01-30:
Adding some images for clarity.

Here is a cup that is sized 3.5 units with a standard 42 mm pitch.
[image]

Here is the same size cup 3.5 units with a half pitch base.
[image]

closed 2025-02-17 by ostat