ostat/gridfinity_extended_openscad #118: Feature request: "offset" for width, depth dimensions

dorianclay, 2025-01-31:
For the [Gridfinity Basic Cup](https://docs.ostat.com/docs/openscad/gridfinity-extended/basic-cup/), please add an option that will allow offsetting the grid dimensions by a given amount.

The existing feature which allows dimensions to be fractional units, such as `width=1.5` and `depth=3.6`, is very useful when fitting Gridfinity into spaces that are not a multiple of 42mm (such as pretty much every drawer). However, this means that subsequent cups might be positioned such that no edge is aligned with the grid, in at least one dimension.

For example, the following basic cup would need `width=3, offset=0.5` and `depth=2.5, offset=0`.

[image]

ostat, 2025-01-31:
Hi, this would actually be quite complicated to do. That's not to say it can't be done, just that I don't know if it will be simple. 

It's not clear to me where you might use it. In the example given I would have have a fractional sized bin on the edge and then the rest of the bins would be whole bins. 
If there was a 0.6 extra on one side, I might create a 2.6 or 3.6 size bin. The next bin would be whole numbers. This helps fill the space/drawer. But the bin is not are reusable.  Having more odd sized bins would become difficult to manage. 

It's worth mentioning there is also the half pitch grid for the bins.

dorianclay, 2025-02-01:
Thanks for the quick commentary! Can you elaborate on what you mean by

> there is also the half pitch grid for the bins.

In my use case, the need for offset bins is due to the combination of drawer size and intended contents; a larger, fractional-sized bin is too big for the object it is meant to contain, and a bin aligned directly over the grid would result in empty space. The example above was meant for illustrative purposes only. If it would help, I can also generate a diagram of the entire drawer layout.

ostat, 2025-02-01:
The base of the bin is 42mm, but you can enable the "half pitch" base that will create a 21mm grid. Its not as flexible as you are after, but it does allow you to offset a bin by half a grid while still keeping it compatible with normal sizing.

This is a 3.5x2 regular base

[image]

This is it with half pitch.

[image]

dorianclay, 2025-02-01:
Seems like that will work perfectly for half units. Let me give that a try.

dorianclay, 2025-02-02:
`half pitch` is more than flexible enough for the use case I've outlined above, thanks @ostat!

closed 2025-02-02 by dorianclay