ostat/gridfinity_extended_openscad #119: [Feature Request] Document intended usage for invoking library directly from scad files

ragaskar, 2025-02-11:
Thanks for building this library! I am finding it quite useful. 

That said, I am having some trouble understanding how one might "correctly" use the library modules from OpenScad scripts (vs just modifying "Customizer" params).

I am using gridfinity to organize a drawer, and in a few places I need the stored item to go all the way to the drawer bottom, that is -- the 5.4mm of the baseplate adds enough height that the item won't fit. 

I initially solved this problem by "hard-coding" spots where I needed to do this: for example, one of these items fits within a 3x3 spot on a larger 5x6 grid. I simply created the standard 5x6 grid, and for the 3x3 section, I `union` that grid with a 3x3 cube, and then `difference` it with the cutout I need. 

Unfortunately, with this approach the iteration is quite tedious and expensive. If I get the positioning wrong on the "hardcoded" portions, I have to print the entire thing again. What I'd like to be able to do is just print the "hardcoded" piece over, and not have to re-print the base at all (as one would expect from the modular system). Even when I have a mostly "final" design, this would allow for future iteration if needed.  

I _could_ use the "custom" grid layout and make the "hardcoded" bit a separate section, but I'd like to 1) get a little bit of added stiffness from printing the entire grid as a single piece, and 2) take advantage of the magnet cutouts to keep these larger sections in place vs just relying on them fitting tightly into the drawer itself. So, for example, to solve the problem I described above, I'd have a 5x6 grid with one 3x3 section that has no "grid edges", just the magnet holders at the corners.

I _can_ emit a "normal" grid and then delete out the sections I don't need in the 3x3 portion, but this results in artifacts where the prior grid box edges would have appeared. I would like to try to emit a base plate with an arbitrary grid unit size, and use that instead. The underlying modules seem mostly to allow this, but given the way the includes/use-s are configured in the modules dir seems to prevent this type of usage. For example, module_gridfinity.scad includes the gridfinity_constants.scad and then includes the functions_gridfinity.scad, which I believe will force (e.g.) CalcDimensionWidth to always use a gf_pitch of 42. 

Anyways, I guess I'm trying to figure out if there's an easy way to do this that perhaps I'm missing -- like, perhaps I can just set the gf_pitch, then re-include the gridifinity_functions file, and that will redefine CalcDimensionWidth such that it uses my custom gf_pitch. Are usages of this func in the modules closures? I don't know, I haven't written much complex behavior yet with OpenSCAD, so I am unaccustomed to idiomatic usage -- this is where docs around intended usage would be helpful -- even if the docs state "Only customizable vars can be changed" -- that would help set my expectations around what I can do without making major changes to the library.

ostat, 2025-02-11:
There is currently no way to do this. Even with the custom grid would be tough...

I have not documented the script and how it works.  That is not to say you cant use to to make custom changes. I have just not documented it. I will try look in to that. 

For your problem, I think a good choice would a flat based bin. This does not have the grid in the middle of the bin, just around the sides.
Recently I made it so the bases could be clipped together. So I am thinking we could make a single custom cell that you could then clip to the other bases.
For example say you make a 4x3 flat base bin. You then make a single 4x3 grid outline. This could then be clipped to other grids.

Thoughts?

Regarding the `gf_pitch` being hard coded, gf_pitch being fixed is an assumption made it a lot of places. I am working on changing that. but as you can imagine that is a bit of a project.... There are a lot of assumptions.

ostat, 2025-02-11:
I have added the option to change the pitch. It allows for changing the pitch from (42 and 7), as well as having different pitch for x and y.
I'm sure this will have bugs, as there are a lot of changes and broken assumptions.

The changes are in this branch if you are interested.
https://github.com/ostat/gridfinity_extended_openscad/tree/custom_pitch

[image]

ragaskar, 2025-02-12:
oh wow, thank you. I did not expect changes to the code from this! Especially since the gf_pitch seems like a quite difficult value to parameterize. 

I had much less luck given my lack of familiarity with the code, and wound up deciding I was letting perfect stand in the way of good and wrote a coarse "wrapper" that "cuts out" a larger hole in the grid ([wrapper gist](https://gist.github.com/ragaskar/d45b2e91016ddd7fc3820434592c2c41)). (One surprise others taking this approach might encounter -- it is necessary, at least according to the intellij openscad plugin I am using -- to remove the customizer comment directives, otherwise the syntax checker throws errors, as it appears to want to validate these and does not like them inside a module). 

I finished a print overnight and so far it seems to work well enough for my purposes? I definitely think that having a programmatic API to the modules is beneficial (or at least satisfies my aesthetic sense of being able to easily see the assemblage of elements). There's still a lot of opportunities for cleanup in the calling code, but here's what a sample invocation looks like: 

[image]

I will definitely try out the branch, thank you!!!! That will help me clean up the grid crossbar elements. I suspect for now the wrapper is my best option, but long-term I could imagine a re-write that teases apart some of the modules and would allow for well, modular usage. I'm not sure my needs are sufficient driver for me to take on that work myself, but you never know, seems like a good way to learn OpenSCAD (I've already learned quite a bit by creating the simple wrapper). 

In either case, I think it makes sense to close this issue -- my overall question re: usage seems to be answered (usage is primarily expected to be via Customizer, and that is technically self documenting). 

Thanks again!

ostat, 2025-02-12:
I would be keen to see you print and how you use it. 

In the file 'module_gridfinity.scad' there is a function called pad_oversize. This is what creates the negative volume that is subtracted from the base plate. 

This would produce clean edges. You can create a normal baseplate then in you custom code render and position a pad of your desired size. 

I agree documenting the code is a good idea.