ostat/gridfinity_extended_openscad #123: Feature request: Custom cutlery

mmalmeida, 2025-02-23:
I saw this customisable cutlery organiser for gridfinity:
https://www.printables.com/model/267303-customized-gridfinity-silverware-holder

I think this would be an awesome addition to your library.
It seems not very different from other models you already have (like the holes one).

I also see the potential to have more usages than just cutlery.

ostat, 2025-02-24:
I have an updated version of that locally for Gridfinty Extended.  However I was never happy with how it turned so have not pushed the changes.
I changed it to use dividers instead of cutouts. As this uses less plastic, and in my opinion looks cleaner.
Otherwise the cavity sizes are essentially the same I believe. I was just never happy with how the result looked.

[image]

mmalmeida, 2025-02-24:
That looks quite interesting actually (and I agree with you, less *plastic* is more). But I understand what you mean about not looking incredible.
I am willing to give it a go an make some tests if you publish it on makerworld - can you do it?

mmalmeida, 2025-02-24:
Step 1: publish this (I will test it).
Step 2 (crazy idea): combine this with the recently published "removable wall dividers" - make these walls removable as well :)

Monniasza, 2025-02-24:
@ostat You can create a 2d array of utensils for unlimited utensils.

ostat, 2025-02-24:
> @ostat You can create a 2d array of utensils for unlimited utensils.

The code supports a list of Utensils, where you could specify any number you wanted. You can't have an u limited list in the customiser, the ui won't support it. 
Do you mean something different?

ostat, 2025-02-25:
This is what I had in mind for the removable dividers.
You can enter the number of slots to span. Then you can pick what walls to print. You cant adjust the bend position or angle.

[image]

mmalmeida, 2025-02-26:
This looks nice - I saw your post on reddit and will try it out.
One immediate practical issue/question: how do you recommend one mimics the "wide/narrow" distance of the other approach?
i.e I understand that the "bend" will be the number of lateral "slots" (which are based on the thickness of the removable walls), but without having tried I don't see how you can translate the "ok this spoon has 5/1cm  wide/narrow, so I need a bend of x".
What do you advise?

ostat, 2025-02-28:
> What do you advise?

The two solutions solve the problem in quite different ways. There are pros and cons depending on what you would like

Option 1: is very flexible in the layout and placement of the walls. However once its printed its fixed. So you can measure everything and get it exactly as you might like it, however once its printed its done. If you want to change it you need to reprint it.

Options 2: The removable walls, is less flexible in the position and layout of the walls, as you need to pick a set spacing. But it allows for adjusting after printing. 

If I was to use option two, I would decide on my bin size, then pick my wall spacing and thickness. I might like 3mm thickness of the walls, 10mm or 15mm spacing between them.
After I print my bin, I would layout the utensils, and print what ever walls seems appropriate.  If the walls is not quite right (I printed a two step and I want a three step), or I later change my mind in the design I would just print another wall with a different step over. 

>Step 2 (crazy idea): combine this with the recently published "removable wall dividers" - make these walls removable as well :)
The removable walls cant be infinity configurable. There needs to be fixed set spacing. So listing the utensil for this approach does not make sense. Maybe I'm miss understanding.

mmalmeida, 2025-03-16:
I'm playing with this today.
Here are a couple suggestions for the help labels on the parameterisation - the current text doesn't make it clear what each parameter does:

**divider wall thickness** : Thickness/protrusion to the inside of the box of the divider wall
**divider wall spanning**: Number of slot spanning dividers to generate. Walls will be generated with increasing spanning up to the selected level (e.g 3 will generate 3 walls, one with spanning 1, one with spanning 2 and one with spanning 3).

**Divider clearance** - I don't yet understand what this one does :)

mmalmeida, 2025-03-17:
@ostat - can you explain what Divider clearance does?

and also, is the base of the cup supposed to be printed with stripes (like slots for the walls to rest in? if so, what controls the depth of these stripes?

mmalmeida, 2025-03-17:
and also, is it possible to add a parameter to tell where in the wall the bend should be?
It could be absolute units or from 0-1.
e.g. 0.5 the curve is at half the wall (so if the wall has size 2, bend will be at 1) - this seems to be the current behaviour
0.9: the curve will be almost at the top

ostat, 2025-03-30:
Sorry for the delay in response. 
> **divider wall thickness** : Thickness/protrusion to the inside of the box of the divider wall 

The **divider wall thickness**  is the thickness of the actual wall, not how far the support comes in to the bin.

**divider wall spanning**: Number of slot spanning dividers to generate. Walls will be generated with increasing spanning up to the selected level. (e.g 3 will generate 3 walls, one with spanning 1, one with spanning 2 and one with spanning 3).

I like your suggestion, I will update, thanks for the input.

> **Divider clearance** - I don't yet understand what this one does :)

Is the clearance between the slot and the removable wall. For example, if you have `divider wall thickness = 2` the wall and the slot will be 2mm. If the clearance is 0.2, the slot will stay at 2mm and the wall will reduce to 1.8mm. This allows for a nicer fit. The reason I reduce the wall and not the slot is because its easier to reprint the walls over the bin. There are two value one is the width of the wall one is the length.

> and also, is the base of the cup supposed to be printed with stripes (like slots for the walls to rest in? if so, what controls the depth of these stripes?

Yes, the slots wont cut in to the cup wall, instead the divider wall support thickness would be added to those walls. You can increase this thickness. For example, say the cup has 1.5mm walls, and you add removable slots with 2mm supports. The outer wall to inner support is 3.5mm, the depth of the slot is 2mm.

> and also, is it possible to add a parameter to tell where in the wall the bend should be?
> It could be absolute units or from 0-1.
> e.g. 0.5 the curve is at half the wall (so if the wall has size 2, bend will be at 1) - this seems to be the current behaviour
> 0.9: the curve will be almost at the top

Good idea, will add.